littleyoda/ha-pysmaplus #2: New type of SMA-inverters

frederikvoGIT, 2024-05-23:
Hi Littleyoda,

I just want to let you know SMA is starting to distribute a new range of inverters running with Ennexos. (Sunny Boy Smart Energy)

I'm pretty new in the Github- and self-programming-world but wanted to notify you so you can maybe add it in your explanation.

I'm going to try your Pysmaplus next days. So keep fingers crossed.

Best regards,
Frederik

littleyoda, 2024-05-23:
Thanks for the tip.

The integration behavior you described in the HA forum fits in perfectly with this
Please send me the diagnostic information. Then I should be able to integrate the support. 

For reference: 
https://community.home-assistant.io/t/support-for-more-sma-devices-e-g-tripower-x/713737/21

littleyoda, 2024-06-10:
Thanks for your help [frederikvoGIT](https://github.com/frederikvoGIT)

closed 2024-06-10 by littleyoda