littleyoda/ha-pysmaplus #4: Problem with all entity of inverter

vindiesel891, 2024-06-12:
Hi,

nine Hours ago home Assistant show me this problem:
Can you see the attached Photo.
If I click on an entity i receive this message: this entity is no longer provided by the pysmaplus integration. 

[image]

Can you risolve the problem please?
Thanks

littleyoda, 2024-06-12:
Can you please send me the diagnose information?

https://littleyoda.github.io/

[image]

vindiesel891, 2024-06-12:
With the smartphone i’m not able to download the diagnostic file. Is it
normal?
Thanks

vindiesel891, 2024-06-12:
This is the diagnostic file!

littleyoda, 2024-06-12:
Unfortunately, the diagnostics function is probably defective in the current Home Assisant version. Since I can't do much without the information, we'll probably have to wait for a new, corrected version of HA.

If the historical data is not important, you can also try deleting the device and adding it again. Please make a backup first.

ulrich379, 2024-07-09:
Morning, may be i have the same problem. If i make an reboot of the system (hoass), the network is confused. In the Logviewer i have this Message every 2 seconds: WARNING (Mainthread) (pymodbus.logging) Failed to connect. If i recover the backup made before restart everything is fine.

littleyoda, 2024-07-09:
> Morning, may be i have the same problem. 
I created a new Issue because as this are probably different problems, as your problem occurs with Modbus (SHM2 with Grid Guard Code).

littleyoda, 2024-07-09:
[vindiesel891](https://github.com/vindiesel891)
Can you try to create a diagnosic file f the problems persist?

ulrich379, 2024-07-09:
Sorry, vor late answer. I have an X20 and an EMeter20, no SHM2.
I have an Diagnostic File downloaded after Restart.

ulrich379, 2024-07-09:
I think In the first file debugging wos not aktivatet.

Greeting Ulrich Maurer

littleyoda, 2024-07-09:
Can you please send me the diagnosic information bei Mail?
https://littleyoda.github.io/

Sending it to github via mail does not work.

littleyoda, 2024-07-25:
This bug should be fixed by the last release

closed 2024-07-25 by littleyoda