littleyoda/ha-pysmaplus #5: Able to change settings

KimCarlsen, 2024-06-16:
hi

I was wondering if it is possible to alter the device settings in the inverter?
It would be nice to alter the feed-in settings, to be able to limit export, when prices goes negative.
Like this https://my.sma-service.com/s/article/Tech-Tip-How-to-Setup-Zero-Export-with-only-an-SMA-Energy-Meter?language=en_US

littleyoda, 2024-06-16:
To be on the safe side: You like to temporarily achieve the effect of Zero Export (https://files.sma.de/downloads/ZeroFeedIn-PL-en-11.pdf)?

First of all: I am happy to help provide the "tool", but other people have to test it. As I receive a fixed price per kWh, a zero export is counterproductive for me. That's why I don't know enough about which way is the most promising in the end.

There are several options for regulation/control:
Modbus via inverter
Modbus via Sunny Home Manager 2
Setting via the local website of the inverter
Setting in the Sunny Home Portal (as described in the file above) 

The topic is not easy at the moment. With the latest changes to the SHM2 firmware, the settings in the inverter are partially overwritten by the SHM2, so that you have to control the SHM2 via Modbus. This solution probably regulates with greater latencies. You also need a personalised Grid Guard code from SMA.

Irrespective of this, I have built in an option to adjust the power limit for the enneox-based systems. This is technically the same way as when you enter the value on the inverter's website.
@frederikvoGIT // https://github.com/littleyoda/ha-pysmaplus/issues/2 tested this method in practice and had massive problems with the inverter no longer accepting the changed values.

I could build in similar support for webconnect, but for this I need the support of someone with a webconnect device and some technical understanding. (Linux system and knowledge would be great)

KimCarlsen, 2024-06-16:
Yes exactly.
My prices are calculated by the hour, based on prices on the nord pool trading exchange. where i get the day ahead prices [nordpool](https://data.nordpoolgroup.com/auction/day-ahead/prices) and they are from time to time negative.
So idea is to activate, automatically from HA, zero export when the prices are below 0.
I have a SMA STP 8.0 with an energy meter connected, currently connected with Webconnect. (using pysmaplus).
I'm gladly help in testing, but i'm not a linux or coding guy.

KimCarlsen, 2024-06-16:
don't know if it possible, , but i've opened a supportticket at SMA

frederikvoGIT, 2024-06-16:
Did some testing tonight, reprogrammed the automation and now got it working when i manualy triggered the automation. (don't know why it refuses to work via the normal automation)

_One automation to start dynamicly limit:_
alias: Begrenzing PV
description: ""
trigger:
  - type: power
    platform: device
    device_id: 2cdf0157cb62ace72eaddbb0d372f7d1
    entity_id: cb54a21d6efd8b1371ac880bb07d9b0d
    domain: sensor
    above: 0
condition:
  - condition: numeric_state
    entity_id: sensor.nordpool_kwh_be_eur_3_10_021
    below: 0
action:
  - service: pysmaplus.set_value
    metadata: {}
    data:
      entity_id: sensor.sunny_boy_smart_energy_5_0_inverter_power_limit
      value: "{{states(\"sensor.grenswaarde_omvormer\")}}"
mode: restart

_One automation to end the dynamic power limit:_
alias: Opheffen  begrenzing PV
description: ""
trigger:
  - type: power
    platform: device
    device_id: 2cdf0157cb62ace72eaddbb0d372f7d1
    entity_id: cb54a21d6efd8b1371ac880bb07d9b0d
    domain: sensor
    above: 0
condition:
  - condition: numeric_state
    entity_id: sensor.nordpool_kwh_be_eur_3_10_021
    above: 0
action:
  - service: pysmaplus.set_value
    metadata: {}
    data:
      entity_id: sensor.sunny_boy_smart_energy_5_0_inverter_power_limit
      value: 4997
mode: restart

frederikvoGIT, 2024-06-16:
> Did some testing tonight, reprogrammed the automation and now got it working when i manualy triggered the automation. (don't know why it refuses to work via the normal automation)
>
> _One automation [...]

Think i found it (attribute was not assigned).
Needs to test it tomorrow.

frederikvoGIT, 2024-06-17:
Found it and got everything working on my Sunny Boy Smart Energy 5.0, with following automation:

- id: '1718466655761'
  alias: Begrenzing PV
  description: ''
  trigger:
  - type: monetary
    platform: device
    device_id: dc390a61e0974f3806762571776d1783
    entity_id: cbd64cb93eacd39b7044bb46fcb87aa6
    domain: sensor
    below: 0
  condition:
  - condition: numeric_state
    entity_id: sensor.sunny_boy_smart_energy_5_0_power_l1
    above: 0
  action:
  - repeat:
      sequence:
      - service: pysmaplus.set_value
        data:
          entity_id: sensor.sunny_boy_smart_energy_5_0_inverter_power_limit
          value: '{{states("sensor.grenswaarde_omvormer")}}'
      while:
      - type: is_monetary
        condition: device
        device_id: dc390a61e0974f3806762571776d1783
        entity_id: cbd64cb93eacd39b7044bb46fcb87aa6
        domain: sensor
        below: 0
  mode: restart
- id: '1718486166803'
  alias: Opheffen  begrenzing PV
  description: ''
  trigger:
  - type: monetary
    platform: device
    device_id: dc390a61e0974f3806762571776d1783
    entity_id: cbd64cb93eacd39b7044bb46fcb87aa6
    domain: sensor
    above: 0
  condition:
  - condition: numeric_state
    entity_id: sensor.nordpool_kwh_be_eur_3_10_021
    above: 0
    attribute: current_price
  - type: is_power
    condition: device
    device_id: 2cdf0157cb62ace72eaddbb0d372f7d1
    entity_id: cb54a21d6efd8b1371ac880bb07d9b0d
    domain: sensor
    above: 0
  action:
  - service: pysmaplus.set_value
    metadata: {}
    data:
      entity_id: sensor.sunny_boy_smart_energy_5_0_inverter_power_limit
      value: 4999
  mode: restart

KimCarlsen, 2024-06-17:
wow, that was fast :-) thanks a lot.
I'll try it out and waiting for negative prices to see how it flows. tomorrow is all above 0

littleyoda, 2024-06-17:
On remark: [frederikvoGIT](https://github.com/frederikvoGIT) Solution works only with a "enneox-OS" based Inverter.

Another solution using modbus:
https://community.home-assistant.io/t/modbus-data-from-sma-inverter/122425/314?u=swante

frederikvoGIT, 2024-06-17:
Yeah, i did the test with temporarily setting the limit on 0,2€/kWh. 😉

And  now indeed very high prices tomorrow. 
Let's hope for negative prices next weekend again.

KimCarlsen, 2024-06-18:
so it does not work with an STP inverter?

littleyoda, 2024-06-18:
> so it does not work with an STP inverter?

correct.
But you can try the modbus solution.

frederikvoGIT, 2024-06-18:
> damn. Modbus might be doable, but not as easy. Then i suggest remove the services i HA, if the connection method doesn't allow it. Otherwise i bet there will be more issues on this.

But @littleyoda please keep the services for the SB SE-series.🙏

littleyoda, 2024-06-18:
Removing the Service does not make sense at all. 
Just because it's not supported on a device doesn't mean I'm going to remove it completely.

Before you use the service call "set parameter", you should first use the service call "get values range" to determine whether you can change the sensor. However, I have just realized that the error message from "get value range" is not particularly informative. I'll have to change it.

The topic is also documented:
https://github.com/littleyoda/ha-pysmaplus/blob/main/docs/set_parameter.md

I am already working on supporting the SHM2 via Modbus. This can be only used, if you have a Grid Guard code from SMA. This is particularly relevant for people who still want to have battery control after the changes to the firmware.

I may integrate Modbus support for other devices for control issues.  I don't know yet if and when something will come, as I'm still working on the SEMP (Simple Energy Management Protocol) and I would like to finalise it.

I'm not sure right now whether a non-modbus control system for webconnect devices makes any sense at all. Perhaps someone can tell me whether there are relevant parameters that can be controlled via the web interface but not via modbus.

I would also appreciate information on which Modbus addresses are relevant at all.

KimCarlsen, 2024-06-19:
I wasn't saying to remove it completely, but perhaps disable it, if a connection method that don't support it is chosen.
Let me know if i can help with testing webconnect. I will try with modbus when there is time for it.

frederikvoGIT, 2024-06-19:
@littleyoda FYI, by the end of this year i might extend my installation with a small battery on my SB SE 5.0, wich i only want to charge and discharge at 10A (480W) maximum.
But to my surprise i see that the parameters "Battery Batt. manuf's max. charge capac." and "Batt. manuf's max. disch. capac." are not editable in the Ennexos Sunnyportal. 
Am i missing something?

[image]

frederikvoGIT, 2024-06-25:
> @littleyoda FYI, by the end of this year i might extend my installation with a small battery on my SB SE 5.0, wich i only want to charge and discharge at 10A (480W) maximum. But to my surprise i see that the parameters "Battery Batt. manuf's max. charge capac." and "Batt. manuf's max. disch. capac." are not editable in the Ennexos Sunnyportal. Am i missing something?
> 
> ![afbeelding](https://private-user-images.githubusercontent.com/144563140/341180875-e7c1ca3e-76b5-4b2b-892c-08bbf3a41bbb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTkzMDQ0MzAsIm5iZiI6MTcxOTMwNDEzMCwicGF0aCI6Ii8xNDQ1NjMxNDAvMzQxMTgwODc1LWU3YzFjYTNlLTc2YjUtNGIyYi04OTJjLTA4YmJmM2E0MWJiYi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYyNVQwODI4NTBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xOTg4YWY1YWJlNzJlODY2OWQxZDJjODU5ZDUzMDk3ODU4NWViMDZjYjE3ZGRlYTIzZDhjNmFhMGVmNjRmZTA4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.1GzK74vWw8OfKFf9md4CM9AjXXWcUk1_LQP8w0P76AQ)

Kind reminder.

frederikvoGIT, 2024-06-25:
> @frederikvoGIT @littleyoda is doing this in his spare time for free. Such comments do net help. I am sure @littleyoda has not forgot it, he just did not have time to look after it.

I had a good month ago conversations on almost a daily base (sometimes on hourly base).
But you are right, he could be occupied at the moment.

I apologise to @littleyoda .

littleyoda, 2024-06-26:
My understanding is that this value specifies the maximum and not the " preferred" charging/discharging speed.

Since you do not have a battery connected, it may well be that the relevant settings for this topic are missing.

Until someone reports the diagnostic information from a Sunny Boy Storage with connected batteries, I can't say anything.

frederikvoGIT, 2024-06-26:
> My understanding is that this value specifies the maximum and not the " preferred" charging/discharging speed.
>
> Since you do not have a battery connected, it may well be that the relevant settings [...]

Ah ok, thanks.

I'll look for a temporarily batterie.

littleyoda, 2024-07-25:
Closed. 
Feature Request "Change Parameters via Webconnect" can be found in Issue  #10

closed 2024-07-25 by littleyoda

frederikvoGIT, 2024-08-06:
Hi Sven,

A friend of mine (Bart Jocque in CC) has a SMA Sunny Tripower Smart energy 10.0 (not Enexos) attached to 3X SMA Home Storage batteries and the SMA EVcharger (both Enexos together with his Sunny Portal).
And is running Home Assistant as a monitoringtool. (for the actual management of the energy he uses the Sunny Home Manager)

Would it be of any help to check how the prefered charging/discharging speed works on these systems if he shows his loggings?

Vriendelijke groeten,
Frederik Van Overloop

frederikvoGIT, 2024-08-06:
FYI; SMA published a video on youtube on which they ask to pose questions if you have them.
So that's what i did ;-)
https://www.youtube.com/watch?v=tmdXhnflGXM&lc=UgzkNA_YRDxCSCDE6YF4AaABAg.A6nMf_agYxVA6nbtBDO0EH

Vriendelijke groeten,
Frederik Van Overloop

frederikvoGIT, 2024-08-13:
In the meantime i:
-posed the question in the serviceportal https://my.sma-service.com

-found a a paramter "fallback of the limitation of the battery"
[image]