littleyoda/ha-pysmaplus #6: Error occurred loading flow for integration pysmaplus

Grizzler89, 2024-06-28:
Used Version of Home Assistant: 2024.6.4
Used Version of ha-pysmaplus: v0.3.1

My three integrations stopped working. No more value updates.
- SMA Home Manager 2.0 (speedwireem)
- SMA SUNNY TRIPOWER SE 8.0 (webconnect)
- SMA SUNNY TRIPOWER 8.0 (webconnect)

I removed all three integrations, reinstalled pysmaplus, restartet HA, rebootet the VM.

Logger: homeassistant.setup
Quelle: setup.py:320
Erstmals aufgetreten: 17:40:54 (1 Vorkommnisse)
Zuletzt protokolliert: 17:40:54
Setup failed for custom integration 'pysmaplus': Unable to import component: cannot import name 'ModbusException' from 'pymodbus' (/usr/local/lib/python3.12/site-packages/pymodbus/__init__.py)

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 992, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1052, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 131, in protected_loop_func
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/pysmaplus/__init__.py", line 9, in <module>
    import pysmaplus as pysma
  File "/usr/local/lib/python3.12/site-packages/pysmaplus/__init__.py", line 17, in <module>
    from .device_shm2 import SHM2
  File "/usr/local/lib/python3.12/site-packages/pysmaplus/device_shm2.py", line 7, in <module>
    from pymodbus import ModbusException
ImportError: cannot import name 'ModbusException' from 'pymodbus' (/usr/local/lib/python3.12/site-packages/pymodbus/__init__.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 320, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1012, in async_get_component
    self._component_future.result()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1004, in async_get_component
    comp = self._get_component()
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1052, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 131, in protected_loop_func
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/pysmaplus/__init__.py", line 9, in <module>
    import pysmaplus as pysma
  File "/usr/local/lib/python3.12/site-packages/pysmaplus/__init__.py", line 17, in <module>
    from .device_shm2 import SHM2
  File "/usr/local/lib/python3.12/site-packages/pysmaplus/device_shm2.py", line 7, in <module>
    from pymodbus import ModbusException
ImportError: cannot import name 'ModbusException' from 'pymodbus' (/usr/local/lib/python3.12/site-packages/pymodbus/__init__.py)

littleyoda, 2024-06-28:
How was your HA installed?
[Installation-methods](https://www.home-assistant.io/blog/2020/05/26/installation-methods-and-community-guides-wiki/#supported-installation-methods)

Can you tell me which version of pymodbus you have installed?
`pip list pymodbus|grep pymodbus
`

Grizzler89, 2024-06-28:
It's the first mehtod: "Home Assistant". Running als VM in ProxmoxVE.

~ # pip list pymodbus
Package    Version
---------- -------
awake      1.0
packaging  24.0
pathspec   0.12.1
pip        24.0
pulsemixer 1.5.1
pyparsing  3.1.2
PyYAML     6.0.1
setuptools 69.5.1
yamllint   1.35.1

Grizzler89, 2024-06-28:
Now it works again.

What i tried:
- Restoring old Versions of HA Core.
- Installing pymodbus using pip install pymodbus

```
~ pip list            
Package    Version
---------- -------
awake      1.0
packaging  24.0
pathspec   0.12.1
pip        24.0
pulsemixer 1.5.1
pymodbus   3.6.9 <-- latest version. I don't know if it's the right one now
pyparsing  3.1.2
PyYAML     6.0.1
setuptools 69.5.1
yamllint   1.35.1
```

Now i'm on HA Core 2024.6.4 (again) and pymodbus is installed.
pysmaplus is downgraded to 0.2.14. Probably happend when restoring a backup of HA Core.

Do you have any idea what went wrong?

littleyoda, 2024-06-28:
I am currently preparing a new version that change the pymodbus requirements to >=3.6.9.
This will then match your feedback.

littleyoda, 2024-06-28:
v0.3.6 released. Please test it with this version.

Grizzler89, 2024-06-29:
Works fine. Thank you!

closed 2024-06-29 by littleyoda