littleyoda/ha-pysmaplus #7: New Energy Sensors for Tripower X 15

pkaempf, 2024-07-08:
EnnexOS-based SunPower Tripower X 15, Firmware version 03.06.15.R 

This Sensors would be great to configure the Energy Dashboard.:
Measurement.Metering.DyWhOut: Daily yield
Measurement.Metering.PCCMs.PlntCsmpWh: Energy drawn at the grid connection point
Measurement.Metering.PCCMs.PlntWh: Feed-in energy at the grid connection point

Interesting but less needed:
Measurement.DcMs.TotDcEnCntWh[0]: Energy released by string [A]
Measurement.DcMs.TotDcEnCntWh[1]: Energy released by string [B]
Measurement.DcMs.TotDcEnCntWh[2]: Energy released by string [C]

Peter

littleyoda, 2024-07-08:
Yes, I can implement if I have some time.

However, the purpose of the following two measured values is completely unclear to me.

- Measurement.Metering.PCCMs.PlntCsmpWh: Energy drawn at the grid connection point
- Measurement.Metering.PCCMs.PlntWh: Feed-in energy at the grid connection point

my first assumption was that they are the same as GridMs.TotWhIn and  GridMs.TotWhOut. However, the values on my system do not match. I can't match the values that I get back at all.

```
          "Metering.PCCMs.PlntWh": {
            "name": "Metering.PCCMs.PlntWh",
            "value": 7056176,
            "origname": "Measurement.Metering.PCCMs.PlntWh"
          },
          "Metering.PCCMs.PlntCsmpWh": {
            "name": "Metering.PCCMs.PlntCsmpWh",
            "value": 3053011,
            "origname": "Measurement.Metering.PCCMs.PlntCsmpWh"
          },

          "Metering.GridMs.TotWhIn": {
            "name": "Metering.GridMs.TotWhIn",
            "value": 6135006.8,
            "origname": "Measurement.Metering.GridMs.TotWhIn"
          },
          "Metering.GridMs.TotWhOut": {
            "name": "Metering.GridMs.TotWhOut",
            "value": 13483003.4,
            "origname": "Measurement.Metering.GridMs.TotWhOut"
          },
```

pkaempf, 2024-07-08:
I have a SMA ENERGY METER connected to my TriPower X 15.
I can find:
- GridMs.TotWhIn: Counter reading of power drawn counter
- GridMs.TotWhOut: Grid feed-in counter reading
as Channels of the SMA Energy Meter Device. I do not really understand the descriptions.

Metering.PCCMs.PlntCsmpWh and Metering.PCCMs.PlntCsmpWh are Channels of the entire PV-System.
Im not able to verify the values because the converter was changed and first they installed the energy meter the wrong way. Now everything is messed up a bit.
Does this help??

littleyoda, 2024-07-08:
For reference. 
The orig. name and the German translation of the values:

'Measurement.Metering.PCCMs.PlntCsmpWh': {
'210001': 'Energiebezug am Netzanschlusspunkt',
'210001': 'Energy drawn at the grid connection point',

'Measurement.Metering.PCCMs.PlntWh'
'210000': 'Eingespeiste Energie am Netzanschlusspunkt',
'210000': 'Feed-in energy at the grid connection point',

'Measurement.Metering.GridMs.TotWhIn'
'210076': 'Zählerstand Bezugszähler',
'210076': 'Counter reading of power drawn counter',

Measurement.Metering.GridMs.TotWhOut
'10714': 'Zählerstand Einspeisezähler',
'10714': 'Grid feed-in counter reading',

VedMod, 2025-02-17:
Any update here?
Specially i am interested in the value from Measurement.Metering.DyWhOut, so could you add this maybe first without the others?

In the moment i have to get the daily output via Modbus into HA, but the Total Output i can't get via Modbus, so with the daily output, i could move all my boards to your integration and deactivate my modbus

littleyoda, 2025-02-20:
"Measurement.Metering.DyWhOut" added for next version

closed 2025-02-20 by littleyoda