littleyoda/ha-pysmaplus #8: Modbus connection problems

littleyoda, 2024-07-09:
Morning, may be i have the same problem. If i make an reboot of the system (hoass), the network is confused. In the Logviewer i have this Message every 2 seconds: WARNING (Mainthread) (pymodbus.logging) Failed to connect. If i recover the backup made before restart everything is fine.

_Originally posted by @ulrich379 in https://github.com/littleyoda/ha-pysmaplus/issues/4#issuecomment-2216707778_

littleyoda, 2024-07-09:
I assume you are using the SHM2 with Grid Guard Code.

0. Please enable the debug logging

[image]

2. Try to reload the integration

[image]

3. Can you please test if you are able to connect to your inverter using a modbus program?
e.g. https://qmodbus.sourceforge.net/
Please you there config as shown in the screenshot. But change the IP-Address.
[image]

ulrich379, 2024-07-09:
I have a working Modbus Connection to the invererter:
modbus
 - name: sma_hub
    type: tcp
    host: 10.10.10.173
    port: 502
    sensors: !include sma_adresses.yaml
    delay: 0
    message_wait_milliseconds: 30
    timeout: 5
I have testet with deaktivatet modbus.yaml but the error persist.

grafik.png (view on web)

littleyoda, 2024-07-09:
You wrote:
> Sorry, vor late answer. I have an X20 and an EMeter20, no SHM2. 

I have just lost the overview.

Which interface are you using?

[image]

If you are not using "Sunny Home Manager 2 with Grid Guard Code", your problem has nothing to do with my integration.

The Energy Meter 2 does not support modbus, as far as I know.

We can also continue in german, if it is easier for you.

ulrich379, 2024-07-09:
I use the ennexos interface

 
 

grafik.png (view on web)

closed 2024-08-03 by littleyoda