littleyoda/ha-pysmaplus #10: [Feature Request] Set Parameters via webconnect

littleyoda, 2024-07-25:
https://community.home-assistant.io/t/feature-to-set-power-limit-of-sma-stp-inverter-from-home-assistant/585408/

littleyoda, 2024-09-28:
The next version (0.4.5) will have improved support for parameters when using the webconnect interface.  The feature set should now match that of the enneox interface.
Unfortunately, all untested because I don't have an inverter with webconnect. Test at your own risk.

[image]

[image]

[image]

KimCarlsen, 2024-10-12:
tried it on my STP 8.0 inverter with webconnect.
using plugin version 4.5.1

Trying to set value Active power limitation to 0 and then get the value afterwards where it still shows 8000w

```
2024-10-12 18:08:39.232 INFO (MainThread) [custom_components.pysmaplus] Update pysma 192.168.x.x/webconnect/3xxxxx
2024-10-12 18:08:39.374 DEBUG (MainThread) [custom_components.pysmaplus] Finished fetching pysmaplus data in 0.142 seconds (success: True)
2024-10-12 18:08:43.466 DEBUG (MainThread) [custom_components.pysmaplus.services] Setting 3xxxxx / SN: 3xxxxx Active Power Limitation to 0
2024-10-12 18:08:44.232 INFO (MainThread) [custom_components.pysmaplus] Update pysma 192.168.x.x/webconnect/3xxxxx
2024-10-12 18:08:44.319 DEBUG (MainThread) [custom_components.pysmaplus] Finished fetching pysmaplus data in 0.087 seconds (success: True)
2024-10-12 18:08:49.233 INFO (MainThread) [custom_components.pysmaplus] Update pysma 192.168.x.x/webconnect/3xxxxx
2024-10-12 18:08:49.309 DEBUG (MainThread) [custom_components.pysmaplus] Finished fetching pysmaplus data in 0.076 seconds (success: True)
2024-10-12 18:08:54.233 INFO (MainThread) [custom_components.pysmaplus] Update pysma 192.168.x.x/webconnect/3xxxxx
2024-10-12 18:08:54.386 DEBUG (MainThread) [custom_components.pysmaplus] Finished fetching pysmaplus data in 0.152 seconds (success: True)
2024-10-12 18:08:59.232 INFO (MainThread) [custom_components.pysmaplus] Update pysma 192.168.x.x/webconnect/3xxxx
2024-10-12 18:08:59.308 DEBUG (MainThread) [custom_components.pysmaplus] Finished fetching pysmaplus data in 0.076 seconds (success: True)
```

Let me know if i can assist in troubleshooting

littleyoda, 2024-10-12:
Please check whether you can change the value directly on the inverter's website.
What kind of account are you using?

KimCarlsen, 2024-10-12:
I'm connecting to inverter as installer.

In Inverter I've set those for testing purpose: 
"Active power limitation P, active power configuration" = 345W
"Set active power limit at grid connection point" = 123W

In HA both "Active power limit" and "Inverter power limit" shows 8000W

littleyoda, 2024-10-12:
Can you please send me the diagostic inforamtion with these values?
https://github.com/littleyoda/ha-pysmaplus/blob/main/docs/diagnosticsinformation.md

littleyoda, 2024-10-13:
I checked the diagnostic-file.

In the end, we are talking about different parameters.

<hr>
1.) I have implemented the following, as suggested in the following 
https://community.home-assistant.io/t/feature-to-set-power-limit-of-sma-stp-inverter-from-home-assistant/585408/9

"6802_00832B00", Identifier.active_power_limitation
Inverter.WMax

Interestingly, in your case, only values between 4000 and 8050 are allowed for this parameter.
```
        "6802_00832B00": {
          "9": [
            {
              "low": 4000,
              "high": 8050,
              "val": 8000
            }
          ]
        },
```
<hr>
You have tested:

2.) 6800_0092D70
PCC.WMax
German: Eingestellte Wirkleistungsgrenze am Netzanschlusspunkt  
English: "Set active power limit at grid connection point"
```
{
  "9": [
    {
      "low": 0,
      "high": 500000,
      "val": 123
    }
  ]
}
```

<hr>

3.) 6802_00924200
Inverter.WModCfg.WCnstCfg.W
Germain: "Wirkleistungsbegrenzung in W"
Egnlish: Active power limitation P, active power configuration"
```
{
  "9": [
    {
      "low": 0,
      "high": 8000,
      "val": 345
    }
  ]
}
```

<hr>

It is completely unclear to me why in your case active_power_limit has 4000 as the lower limit.
I'll have to think about what I'm doing. But I'll probably just add your two parameters.

KimCarlsen, 2024-10-14:
Hi

Thanks

Also wierd that apparently there is a active_power_limitation in the inverter. I cannot find that. Don't know what that is.

As I understand it, and what it looks like when I test it.
The "Active power limitation P, active power configuration" is to control the output directly from the inverter.
While the "Set active power limit at grid connection point" is what you export to the grid.

The second one is the one, that is relevant for preventing export when prices are negative. And have some HA automation around this to set it to 0 when prices are below 0 and then max, when positive.

Also I wondered if Kim

littleyoda, 2024-10-15:
I added the following sensor:

"Active Power Limitation (GCP)" in v.0.4.6
webconnect: 6800_0092D70
PCC.WMax
German: Eingestellte Wirkleistungsgrenze am Netzanschlusspunkt
English: "Set active power limit at grid connection point"

littleyoda, 2024-10-15:
Please try the v0.4.6.1. It should be fixed now.
You don't have to remove the old device. The new sensor should be added automatically.

KimCarlsen, 2024-10-15:
Sorry. no new sensors. still 70 entities.
[image]

[image]

littleyoda, 2024-10-15:
strange ... can you please send me a new diagnostic file?

littleyoda, 2024-10-26:
the problem was solved in the latest version.

closed 2024-10-26 by littleyoda