littleyoda/ha-pysmaplus #12: SMA Tripower Einrichtung

alexraven1970, 2024-07-28:
Hi,
ich schreib mal auf deutsch, hoffe das ist nicht gegen die etikette...
Ich bekomm meinen Wechselrichter nicht in HA

**Used Interface/Access Method**
Speedwire: [ x ]

[image]

[image]

[image]

Was mach ich falsch?

littleyoda, 2024-07-28:
Hi

kannst du bitte einmal die Debug-Protokollierung auf der Integrationseite aktivieren?
Danach Home Assistant neu starten
Wenn Home Assistant wieder hochgefahren ist, Debug Protokollierung abschalten.

Die Datei, die dann herruntergeladen wird, bitte einmal an mich schicken.
https://littleyoda.github.io/

[image]

alexraven1970, 2024-07-28:
[home-assistant_2024-07-28T11-03-39.551Z.log](https://github.com/user-attachments/files/16403023/home-assistant_2024-07-28T11-03-39.551Z.log)

Meinst Du das?

littleyoda, 2024-07-28:
Hast du neben den Speedwire-Interface auch noch probeweise den Enneox-Interface getestet? Wenn ja, lösche bitte mal die Integration mit dem Enneox-Interface

Bist du dir bei den Zugangsdaten sicher?

Hast du sie schon mal im Sunny Explorer getestet?
https://www.sma.de/produkte/monitoring-control/sunny-explorer

alexraven1970, 2024-07-28:
[image]
die drei hab ich versucht.
Bein PW bin ich mir sicher, web login funktioniert ja
[image]

alexraven1970, 2024-07-28:
ich seh nur ein interface

alexraven1970, 2024-07-28:
[image]

littleyoda, 2024-07-28:
Wenn du aber einen GUI-Zugriff hast, dann probiere bitte mal den Webconnect Zugang bei dir.

alexraven1970, 2024-07-28:
[image]

klappt nicht

alexraven1970, 2024-07-28:
Terminal? 
ich kann es bedienen wenn ich die korreten Befehle habe ... :-)

littleyoda, 2024-07-28:
Ich würde dich mal ein paar Tage vertrösten wollen.
Ich baue gerade eine Funktion ein, die nach SMA Geräten suchen kann und Information sammelt.
Damit kann ich dir dann hoffentlich besser helfen.

littleyoda, 2024-07-29:
Hi

kannst du bitte mal auf die aktuelle Version 0.3.8 aktualisieren?

Danach unter Eintwicklerwerkzeuge/Dienste einmal den Discovery Service laufen lassen und das Ergebnisse hier posten. (Nachdem du ausführen geklickt hast, siehst du erst mal keine Veränderung. Die Ergebnisse kommen aber nach ca. 20 sekunden. Also einfach warten)
[image]

alexraven1970, 2024-08-01:
Hi,
sorry für de Verzögerung, ich musste die letzten Tage ins Büro. Da fehlen die 3 Stunden Pendeln ... keine Freizeit mehr ... 

okay, update ist durch, keine Fehlermeldungen. Dann wollte ich den Dienst starten, finde nichts unter SMA. Hab den Beehl direkt eingegeben mit dem erwarteten Ergebnis
[image]

alexraven1970, 2024-08-01:
ich lade die repository gerade nochmal, noch ein HA restart, dann versuch ich es nochmal

alexraven1970, 2024-08-01:
taucht in Dienste nicht auf. Und nun?

littleyoda, 2024-08-01:
Kannst du mal unter Integrationen schauen, welche Version installiert ist?

Es sollte 3.8.1 sein.

[image]

[image]

alexraven1970, 2024-08-01:
[image]

alexraven1970, 2024-08-01:
den Integrationseintrag finde ich nur in HACS
über Einastellungen sehe ich keine Integration, nur Gerät und Entität

littleyoda, 2024-08-01:
Füge bitte nochmal wie du schon oben probiert hattest, eine Integration zu. (selbst wenn sie nur einen Fehler zeigt) und probiere es dann nochmal.

alexraven1970, 2024-08-01:
[image]
[image]

alexraven1970, 2024-08-01:
soll ich erst die bestehende löschen?

littleyoda, 2024-08-01:
Ich werde aus deinen Problemen leider nicht schlau.
Es kann sein (kann's gerade nicht testen), dass der Discovery-Service nur verfügbar ist, wenn auch ein Gerät angelegt ist. Ich dachte, es wäre unabhängig davon, ob die Integration laufen oder nicht laufen würde. Ist aber wohl nicht der Fall.
Dann muss ich die Tage nochmal die discovery-Funktion komplett umbauen.

alexraven1970, 2024-08-01:
alles klar, danke dir

littleyoda, 2024-09-21:
There is now a consolidated bug for the problems with Speedwire. #18

closed 2024-09-21 by littleyoda