littleyoda/ha-pysmaplus #13: SBSE 3.6 (EnnoxOS) returns invalid JSON

Lordfritz, 2024-07-29:
Used Version of Home Assistant: 2024.7.3
Used Version of ha-pysmaplus: 0.3.8

Connecting to the inverter works fine, but after a while (couple of hours) no new data is retrieved from inverter. The following log messages appear (refresh time is set to 10 seconds). Restarting the integration fixes the problem for a while.

```
2024-07-29 12:46:57.551 ERROR (MainThread) [pysmaplus.device_ennexos] Request to ttps://192.168.107.5/api/v1/measurements/live did not return a valid json.
2024-07-29 12:46:57.562 ERROR (MainThread) [pysmaplus.device_ennexos] Request to ttps://192.168.107.5/api/v1/parameters/search did not return a valid json.
2024-07-29 12:46:57.562 WARNING (MainThread) [pysmaplus.device_ennexos] Uncommon length of array in parameters request: 0
2024-07-29 12:47:07.547 ERROR (MainThread) [pysmaplus.device_ennexos] Request to ttps://192.168.107.5/api/v1/measurements/live did not return a valid json.
2024-07-29 12:47:07.558 ERROR (MainThread) [pysmaplus.device_ennexos] Request to ttps://192.168.107.5/api/v1/parameters/search did not return a valid json.
2024-07-29 12:47:07.558 WARNING (MainThread) [pysmaplus.device_ennexos] Uncommon length of array in parameters request: 0
2024-07-29 12:47:17.547 ERROR (MainThread) [pysmaplus.device_ennexos] Request to ttps://192.168.107.5/api/v1/measurements/live did not return a valid json.
2024-07-29 12:47:17.559 ERROR (MainThread) [pysmaplus.device_ennexos] Request to ttps://192.168.107.5/api/v1/parameters/search did not return a valid json.
2024-07-29 12:47:17.559 WARNING (MainThread) [pysmaplus.device_ennexos] Uncommon length of array in parameters request: 0
```

Speedwire: [ ]
Webconnect: [ ]
EnnexOS-based: [x]
Energymeter/SHM2: [ ]

Inverter: Sunny Boy Smart Energy 3.6 (firmware v03.08.03.R)
Energymeter:
Sunny Home Manager 2:

littleyoda, 2024-07-29:
can you please active the debug logging and send me the log file after the problem reappeared?

E-Mail can be found here: https://littleyoda.github.io/

[image]

littleyoda, 2024-07-29:
I found the error.

The new version (0.3.8.1) should be available  in HACS within the next 30 min.

Lordfritz, 2024-07-30:
Updated the code and kept it running overnight. The measurements are coming in continuously, so it seems to be fixed. Thanks.

Still get the followimg message once or twice per hour:

`Request to https://192.168.x.x/api/v1/measurements/live did not return a valid json. Code 401`

littleyoda, 2024-07-30:
For now this message is ok.

After login, the Token is only valid for 60(?) minutes. After this time the requests are falling (see your debug message). The libary uses the 401 code as an indicator that a new login is required.

closed 2024-07-30 by littleyoda