littleyoda/ha-pysmaplus #14: With latest version Speedwire does not seem to work with Tripower Smart Energy

nemihome, 2024-08-03:
Used Version of Home Assistant:
Used Version of ha-pysmaplus

**Describe the bug, Remarks, ...**
After update of files integration is not working for a speedwire device

**Used Interface/Access Method**
see https://raw.githubusercontent.com/littleyoda/ha-pysmaplus/main/docs/images/interfaces.png
Speedwire: [X]
Webconnect: [ ]
EnnexOS-based: [ ]
Energymeter/SHM2: [ ]

**List of the integrated SMA devices**
Inverter: Smart Energy Tripower
Energymeter:
Sunny Home Manager 2: X

**Screenshots and Diagnose Information**
If applicable, add screenshots to help explain your problem and send the [Diagnose Information](https://raw.githubusercontent.com/littleyoda/ha-pysmaplus/main/docs/Diagnose-Information.png) to the developer. (E-Mail can be found here: https://littleyoda.github.io/)

```
Traceback (most recent call last):
  File "/opt/homeassistant/lib/python3.12/site-packages/homeassistant/config_entries.py", line 586, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homeassistant/config/custom_components/pysmaplus/__init__.py", line 91, in async_setup_entry
    sma = await getPysmaInstance(hass, entry.data)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homeassistant/config/custom_components/pysmaplus/__init__.py", line 84, in getPysmaInstance
    await sma.new_session()
  File "/opt/homeassistant/lib/python3.12/site-packages/pysmaplus/device_speedwire.py", line 424, in new_session
    raise SmaConnectionException("No connection to device: %s:9522", self._host)
pysmaplus.exceptions.SmaConnectionException: ('No connection to device: %s:9522', '192.168.181.28')
2024-08-03 12:39:24.716 INFO (MainThread) [homeassistant.components.automation.shellies_announce_gen2] Shellies Announce Gen2: Running automation actions
2024-08-03 12:39:24.717 INFO (MainThread) [homeassistant.components.automation.shellies_announce_gen2] Shellies Announce Gen2: Repeating sequence: Iteration 1 of 2 with item: 'shellypro3em-34987a45b630'
2024-08-03 12:39:24.717 INFO (MainThread) [homeassistant.components.automation.shellies_announce_gen2] Shellies Announce Gen2: Repeat at step 1: Running automation actions
2024-08-03 12:39:24.717 INFO (MainThread) [homeassistant.components.automation.shellies_announce_gen2] Shellies Announce Gen2: Repeat at step 1: Executing step call service
2024-08-03 12:39:24.718 INFO (MainThread) [homeassistant.components.automation.shellies_announce_gen2] Initialized trigger Shellies Announce Gen2
2024-08-03 12:39:24.719 INFO (MainThread) [homeassistant.components.automation.shellies_discovery_gen2] Initialized trigger Shellies Discovery Gen2
2024-08-03 12:39:24.719 INFO (MainThread) [homeassistant.components.automation.balkon_pv_aktivieren] Initialized trigger Balkon PV aktivieren
2024-08-03 12:39:24.743 INFO (MainThread) [homeassistant.components.automation.shellies_announce_gen2] Shellies Announce Gen2: Repeating sequence: Iteration 2 of 2 with item: 'proem3-home1-34987a465840'
2024-08-03 12:39:24.743 INFO (MainThread) [homeassistant.components.automation.shellies_announce_gen2] Shellies Announce Gen2: Repeat at step 1: Running automation actions
2024-08-03 12:39:24.743 INFO (MainThread) [homeassistant.components.automation.shellies_announce_gen2] Shellies Announce Gen2: Repeat at step 1: Executing step call service
2024-08-03 12:40:35.677 DEBUG (MainThread) [pysma] Sending POST request to /dyn/getValues.json: {'data': '{"destDev": [], "keys": ["6100_00465700", "6102_00254F00", "6400_00260100", "6380_40451F00", "6100_0046E700", "6180_08412800", "6380_40452100", "6400_0046C300", "6100_00464A00", "6102_40254E00", "6100_40465400", "6100_0046EC00", "6800_00832A00", "6400_00462400", "6100_0046E500", "6100_40464000", "6100_40464200", "6400_00462500", "6100_40466600", "6100_40465300", "6100_0046ED00", "6100_40466500", "6100_00468100", "6100_40466B00", "6180_08214800", "6100_0046C200", "6100_40263F00", "6380_40251E00", "6100_0046E600", "6100_00464800", "6100_00464900", "6100_00664F00", "6100_40463700", "6100_0046EB00", "6180_08416400", "6100_40465500", "6100_40464100", "6180_08414C00", "6100_40463600", "6400_00262200"]}', 'headers': {'content-type': 'application/json'}, 'params': {'sid': 'rNgMO3QdVDGd91QY'}}
2024-08-03 12:40:35.754 DEBUG (MainThread) [pysma] Received reply {'result': {'01B8-xxxxxF8C': {'6100_00465700': {'9': [{'val': 5001}]}, '6102_00254F00': {'9': [{'val': 2227000}]}, '6400_00260100': {'9': [{'val': 1013687}]}, '6380_40451F00': {'1': [{'val': 48760}, {'val': 37160}]}, '6100_0046E700': {'9': [{'val': 23583}]}, '6180_08412800': {'9': [{'val': [{'tag': 569}]}]}, '6380_40452100': {'1': [{'val': 3615}, {'val': 6628}]}, '6400_0046C300': {'9': [{'val': 1039494}]}, '6100_00464A00': {'9': [{'val': 23660}]}, '6102_40254E00': {'9': [{'val': 22}]}, '6100_40465400': {'9': [{'val': 5600}]}, '6100_0046EC00': {'9': [{'val': 0}]}, '6800_00832A00': {'9': [{'low': 10000, 'high': 10000, 'val': 10000}]}, '6400_00462400': {'9': [{'val': 932544}]}, '6100_0046E500': {'9': [{'val': 23340}]}, '6100_40464000': {'9': [{'val': 1344}]}, '6100_40464200': {'9': [{'val': 1357}]}, '6400_00462500': {'9': [{'val': 896325}]}, '6100_40466600': {'9': [{'val': 5712}]}, '6100_40465300': {'9': [{'val': 5600}]}, '6100_0046ED00': {'9': [{'val': 0}]}, '6100_40466500': {'9': [{'val': 5616}]}, '6100_00468100': {'9': [{'val': 5002}]}, '6100_40466B00': {'9': [{'val': 6233}]}, '6180_08214800': {'9': [{'val': [{'tag': 307}]}]}, '6100_0046C200': {'9': [{'val': 4052}]}, '6100_40263F00': {'9': [{'val': 4052}]}, '6380_40251E00': {'1': [{'val': 1763}, {'val': 2463}]}, '6100_0046E600': {'9': [{'val': 23579}]}, '6100_00464800': {'9': [{'val': 23480}]}, '6100_00464900': {'9': [{'val': 23660}]}, '6100_00664F00': {'9': [{'val': 16900}]}, '6100_40463700': {'9': [{'val': 0}]}, '6100_0046EB00': {'9': [{'val': 1290}]}, '6180_08416400': {'9': [{'val': [{'tag': 51}]}]}, '6100_40465500': {'9': [{'val': 5700}]}, '6100_40464100': {'9': [{'val': 1351}]}, '6180_08414C00': {'9': [{'val': [{'tag': 307}]}]}, '6100_40463600': {'9': [{'val': 1497}]}, '6400_00262200': {'9': [{'val': 27383}]}}}}
2024-08-03 12:40:36.882 DEBUG (MainThread) [custom_components.pysmaplus] Finished fetching pysmaplus data in 0.102 seconds (success: True)
2024-08-03 12:40:40.678 DEBUG (MainThread) [pysma] Sending POST request to /dyn/getValues.json: {'data': '{"destDev": [], "keys": ["6100_00465700", "6102_00254F00", "6400_00260100", "6380_40451F00", "6100_0046E700", "6180_08412800", "6380_40452100", "6400_0046C300", "6100_00464A00", "6102_40254E00", "6100_40465400", "6100_0046EC00", "6800_00832A00", "6400_00462400", "6100_0046E500", "6100_40464000", "6100_40464200", "6400_00462500", "6100_40466600", "6100_40465300", "6100_0046ED00", "6100_40466500", "6100_00468100", "6100_40466B00", "6180_08214800", "6100_0046C200", "6100_40263F00", "6380_40251E00", "6100_0046E600", "6100_00464800", "6100_00464900", "6100_00664F00", "6100_40463700", "6100_0046EB00", "6180_08416400", "6100_40465500", "6100_40464100", "6180_08414C00", "6100_40463600", "6400_00262200"]}', 'headers': {'content-type': 'application/json'}, 'params': {'sid': 'rNgMO3QdVDGd91QY'}}
2024-08-03 12:40:40.754 DEBUG (MainThread) [pysma] Received reply {'result': {'01B8-xxxxxF8C': {'6100_00465700': {'9': [{'val': 5002}]}, '6102_00254F00': {'9': [{'val': 2227000}]}, '6400_00260100': {'9': [{'val': 1013693}]}, '6380_40451F00': {'1': [{'val': 48760}, {'val': 37160}]}, '6100_0046E700': {'9': [{'val': 23589}]}, '6180_08412800': {'9': [{'val': [{'tag': 569}]}]}, '6380_40452100': {'1': [{'val': 3640}, {'val': 6679}]}, '6400_0046C300': {'9': [{'val': 1039500}]}, '6100_00464A00': {'9': [{'val': 23610}]}, '6102_40254E00': {'9': [{'val': 22}]}, '6100_40465400': {'9': [{'val': 5700}]}, '6100_0046EC00': {'9': [{'val': 0}]}, '6800_00832A00': {'9': [{'low': 10000, 'high': 10000, 'val': 10000}]}, '6400_00462400': {'9': [{'val': 932546}]}, '6100_0046E500': {'9': [{'val': 23315}]}, '6100_40464000': {'9': [{'val': 1355}]}, '6100_40464200': {'9': [{'val': 1366}]}, '6400_00462500': {'9': [{'val': 896325}]}, '6100_40466600': {'9': [{'val': 5755}]}, '6100_40465300': {'9': [{'val': 5700}]}, '6100_0046ED00': {'9': [{'val': 0}]}, '6100_40466500': {'9': [{'val': 5537}]}, '6100_00468100': {'9': [{'val': 5001}]}, '6100_40466B00': {'9': [{'val': 6281}]}, '6180_08214800': {'9': [{'val': [{'tag': 307}]}]}, '6100_0046C200': {'9': [{'val': 4083}]}, '6100_40263F00': {'9': [{'val': 4083}]}, '6380_40251E00': {'1 [... cut ...] ': 51}]}]}, '6100_40465500': {'9': [{'val': 5700}]}, '6100_40464100': {'9': [{'val': 1362}]}, '6180_08414C00': {'9': [{'val': [{'tag': 307}]}]}, '6100_40463600': {'9': [{'val': 1541}]}, '6400_00262200': {'9': [{'val': 27390}]}}}}
2024-08-03 12:40:41.897 DEBUG (MainThread) [custom_components.pysmaplus] Finished fetching pysmaplus data in 0.118 seconds (success: True)
2024-08-03 12:40:42.684 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry SMADEVPLUS-Speed for pysmaplus
```

littleyoda, 2024-08-03:
Have to tried to reload the integration?

[image]

nemihome, 2024-08-03:
Several times, yes. Restarted HA and deleted the cache.

littleyoda, 2024-08-03:
Can you please send the the  [Diagnose Information](https://raw.githubusercontent.com/littleyoda/ha-pysmaplus/main/docs/Diagnose-Information.png)?
E-Mail can be found here: https://littleyoda.github.io/

nemihome, 2024-08-03:
This is not available which means there seems to be nor more log information.
[image]

I will send you the mail with the standard logs.

littleyoda, 2024-09-21:
There is now a consolidated bug for the problems with Speedwire. #18

closed 2024-09-21 by littleyoda