littleyoda/ha-pysmaplus #15: No connection to Sunny Tripower 4.0 after update

DeltaTango69, 2024-08-25:
Used Version of Home Assistant:2024.8.2
Used Version of ha-pysmaplus:04.1

**Describe the bug, Remarks, ...**
A clear and concise description of what the bug is. 

No connection

**Used Interface/Access Method**
see https://raw.githubusercontent.com/littleyoda/ha-pysmaplus/main/docs/images/interfaces.png
Speedwire: [x]
Webconnect: [x ]
EnnexOS-based: [x ]
Energymeter/SHM2: [ ]

all tried

**List of the integrated SMA devices**
STP 15000TL-30

**Screenshots and Diagnose Information**
[image]

```
2024-08-25 15:38:25 homeassistant  | 2024-08-25 15:38:25.626 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry STP4.0-3AV-40 (Sunny Tripower 4.0) (3012423268) for pysmaplus
2024-08-25 15:38:25 homeassistant  | Traceback (most recent call last):
2024-08-25 15:38:25 homeassistant  |   File "/usr/src/homeassistant/homeassistant/config_entries.py", line 604, in async_setup
2024-08-25 15:38:25 homeassistant  |     result = await component.async_setup_entry(hass, self)
2024-08-25 15:38:25 homeassistant  |              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2024-08-25 15:38:25 homeassistant  |   File "/config/custom_components/pysmaplus/__init__.py", line 91, in async_setup_entry
2024-08-25 15:38:25 homeassistant  |     sma = await getPysmaInstance(hass, entry.data)
2024-08-25 15:38:25 homeassistant  |           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2024-08-25 15:38:25 homeassistant  |   File "/config/custom_components/pysmaplus/__init__.py", line 84, in getPysmaInstance
2024-08-25 15:38:25 homeassistant  |     await sma.new_session()
2024-08-25 15:38:25 homeassistant  |   File "/usr/local/lib/python3.12/site-packages/pysmaplus/device_speedwire.py", line 426, in new_session
2024-08-25 15:38:25 homeassistant  |     raise SmaConnectionException(
2024-08-25 15:38:25 homeassistant  | pysmaplus.exceptions.SmaConnectionException: No connection to device: 192.168.178.10:9522  (3/3)
```

DeltaTango69, 2024-08-25:
Wenn ich "Neu laden" drücke sehe ich kurz das die Verbindung zu 192.168.178.10:9552 nicht funktioniert. Woher kommen die 9552? Wenn ich die Adresse so im Browser eingebe bekomme ich auch keine Verbindung. Nur ohne Port!

littleyoda, 2024-08-25:
Sehe ich es richtig, dass die Verbindung über webconnect funktioniert, "nur" Speedwire nicht funktioniert?

Webconnect läuft über http/s (=> Browser)
Speedwire läuft aber über UDP Port 9552. Deshalb hast du darauf gar keinen Einfluss.

Wenn Webconnect funktioniert, dann bleibt bei Webconnect.

Speedwire sollte eigentlich immer funktionieren. Von daher ist es schon komisch. 
Wenn du eine Linux-Maschine im Netzwerk verfügbar hast, könnten wir uns etwas probieren. Melde dich dann per Mail bei mir. https://littleyoda.github.io/

DeltaTango69, 2024-08-25:
Nein es funktionierte nach dem Update gar keine Verbindung mehr. Vorher schon. Bei mir klappt generell nur die Speedwire Verbindung. Ich habe alles unter Windows WSL2 mit Docker Compose am laufen.

littleyoda, 2024-08-25:
Ok. Das muss ich jetzt etwas sortieren:

- Was funktionierte vorher? (Nur Speedwire?)
- Was funktioniert nun? (Gar nichts? Würde aber nicht zum Screenshot passen)
- Was genau hast du aktualisiert? Meine Integraton? Home Assistant? Weiß du noch die alte Version?

DeltaTango69, 2024-08-25:
Ok, ich versuche Mal Licht ins Dunkel zu bringen  :-)

1) Bei mir funktioniert generell nur die Speedwire Verbindung
2) Vor dem Update funktionierten beide Wechselrichter damit - nach dem Update nur noch der STP-15000TL-30 (siehe Bild oben)
3) Ich habe Homeassistant aktualisiert und dann die HACS Integration von SMA von 0.3.1 auf 0.4.1

littleyoda, 2024-08-25:
Du bist der zweite mit so einem Fehlerbild.
Ich weiß leider aktuell nicht, wo das Problem ist.

Du könntest mit Hilfe von HACS erst mal wieder die 0.3.1 installieren.

tonyhammainen, 2024-09-04:
Hi @littleyoda , continuing the discussion from [home assistant forum](https://community.home-assistant.io/t/support-for-more-sma-devices-e-g-tripower-x/713737/68), as I hit a `"new user cant send more than 3 messages"` limit

1.  I had successfully added the devices with 0.3.4. Upgrading to 0.4.2 didn't break the old ones, so while the device adding may have problems, existing devices stay functional.
2. Running service discovery failed; didn't produce output
3. Trying to add a device fails

I see this error in logs:

```
2024-09-04 13:13:52.921 ERROR (MainThread) [homeassistant.helpers.script.websocket_api_script] websocket_api script: Error executing script. Unexpected error for call_service at pos 1: cannot access local variable 'di' where it is not associated with a value
Traceback (most recent call last):
File "/usr/local/lib/python3.12/site-packages/pysmaplus/device_speedwire.py", line 545, in detect
raise SmaConnectionException("protocol not initialized")
pysmaplus.exceptions.SmaConnectionException: protocol not initialized
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 527, in _async_step
await getattr(self, handler)()
File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 764, in _async_call_service_step
response_data = await self._async_run_long_action(
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 727, in _async_run_long_action
return await long_task
^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/core.py", line 2731, in async_call
response_data = await coro
^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/core.py", line 2774, in _execute_service
return await target(service_call)
^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/pysmaplus/services.py", line 157, in discovery
ident = await identify(r[0], False)
^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/pysmaplus/services.py", line 136, in identify
ret = await pysma.autoDetect(session, url)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/pysmaplus/__init__.py", line 90, in autoDetect
ret = await asyncio.gather(
^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/pysmaplus/__init__.py", line 77, in _run_detect
ret = await sma.detect(ip)
^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/pysmaplus/device_speedwire.py", line 566, in detect
di.status = "failed"
^^
UnboundLocalError: cannot access local variable 'di' where it is not associated with a value
```

littleyoda, 2024-09-12:
Can you please retest the latest version (0.4.3)?
I hope that I have corrected the problems.

littleyoda, 2024-09-21:
There is now a consolidated bug for the problems with Speedwire. #18

closed 2024-09-21 by littleyoda