littleyoda/ha-pysmaplus #16: Sensor grid doesn't wake up

MarkoSchweizer, 2024-09-18:
Used Version of Home Assistant: 13.1
Used Version of ha-pysmaplus: 0.4.2

**Describe the bug, Remarks, ...**

I use the integration for a TripowerX15 and for a STP 15000TL inverter. The TripowerX15 works perfect, but after the night the grip power sensor of the STP15000TL shows "not available" until I reload the configuration. How can I solve it?

**Used Interface/Access Method**
see https://raw.githubusercontent.com/littleyoda/ha-pysmaplus/main/docs/images/interfaces.png
Speedwire: [x ]
Webconnect: [ ]
EnnexOS-based: [ ]
Energymeter/SHM2: [ ]

**List of the integrated SMA devices**
Inverter:  STP 15000TL

**Screenshots and Diagnose Information**
If applicable, add screenshots to help explain your problem and send the [Diagnose Information](https://raw.githubusercontent.com/littleyoda/ha-pysmaplus/main/docs/Diagnose-Information.png) to the developer. (E-Mail can be found here: https://littleyoda.github.io/)

littleyoda, 2024-09-18:
Can you please activate the debug logging for the integration overnight.
And stop the logging tomorrow after sunrise and please send me the logfile compressed by email.
Please also send me the diagnostic information.

closed 2024-11-05 by littleyoda

littleyoda, 2024-11-05:
I'm sorry, but I overlooked your log files.

Unfortunately, they don't help me.

Two approaches:
- In such cases, please test whether you can access the inverter via Sunny Explorer. (https://www.sma.de/produkte/monitoring-control/sunny-explorer)

- I would need a complete recording of the Sunny Explorer network traffic.
(only for advanced users, https://github.com/littleyoda/pysma/blob/master/doc/opentasks.md)