littleyoda/ha-pysmaplus #18: Speedwire does not work in same cases starting with version 0.3.5

littleyoda, 2024-09-21:
Querying Speedwire inverters does not work in certain cases. 

There is currently no bug fix available. The last working version seems to be version 0.3.4.

I have listed  [here](https://www.photovoltaikforum.com/thread/225270-home-assistant-und-sma-ger%C3%A4te/?postID=3948843#post3948843l) (in German) how you can help me with the troubleshooting.

Workaround: downgrade to version 0.3.4

borsti112, 2024-10-09:
I've updated to the Version 4.5.1 but still not able to add the Tripower SE through Speedwire or Webconnect. Invalid authentication.

joshiste, 2024-11-23:
I'm not sure wether this is the correct issue to report this on, but I tried to connect my STP8.0SE (SUNNY TRIPOWER 8.0 SE) using the speedwire protocol integration and it yields the following error:

```
Logger: homeassistant.config_entries
Source: config_entries.py:635
First occurred: 8:57:25 AM (1 occurrences)
Last logged: 8:57:25 AM

Error setting up entry STP8.0SE (SUNNY TRIPOWER 8.0 SE) (sw3015639757) for pysmaplus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 635, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/pysmaplus/__init__.py", line 95, in async_setup_entry
    sma = await getPysmaInstance(hass, entry.data)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/pysmaplus/__init__.py", line 88, in getPysmaInstance
    await sma.new_session()
  File "/usr/local/lib/python3.12/site-packages/pysmaplus/device_speedwire.py", line 434, in new_session
    raise SmaConnectionException(
pysmaplus.exceptions.SmaConnectionException: No connection to device: sma-inverter.fritz.box:9522  (3/3)
```

Afaik the port 9522 is used for the discovery of the Speedwire protocol and I wonder why it is used directly?
According to https://cdn.sma.de/fileadmin/content/www.developer.sma.de/docs/SpeedwireDD-TI-en-10.pdf?v=1699275967 it should be used on the multicast address?

Do let me know if you need additional information.

littleyoda, 2025-11-10:
Ja, richtig. Es unterstützt aktuell nur ein minimal Set an Sensoren. Die anderen werde ich dann nach und nach hinzufügen. Aktuell ist erstmal wichtig, dass es stabil läuft.

Yes, that's right. It currently only supports a minimal set of sensors. I will add the others over time. For now, it's important that it runs stable.