littleyoda/ha-pysmaplus #19: Speedwire and Webconnect:  Invalid authentication.

littleyoda, 2024-10-09:
I've updated to the Version 4.5.1 but still not able to add the Tripower SE through Speedwire or Webconnect. Invalid authentication.

_Originally posted by @borsti112 in https://github.com/littleyoda/ha-pysmaplus/issues/18#issuecomment-2403124712_

littleyoda, 2024-10-09:
I created a new issue, because both speedwire and webconnect are effected in your case.

Can you please test your credentials using  the sunny explorer?
https://www.sma.de/produkte/monitoring-control/sunny-explorer

borsti112, 2024-10-09:
yes, credentials are ok. Login via Webinterface and the sunny-explorer works well.

littleyoda, 2024-10-09:
Just to be sure:
For the Webconnect access, you are using the same access data as for the local website of the inverter.

For the Speedwire access, you are using the access data as in the sunny exploerer (Group: user or installer).

Please take a look at the log file to see if there are any entries for pysma.

littleyoda, 2024-10-09:
And please run the discovery mode:
[image]

borsti112, 2024-10-09:
yes for sure I am using the same login. Both User or installer are not working.

wow, very strange. I tried it a last time to add it via webconnect and it is in. No authentication issue anymore. I don't know why. I really tried it min 10times with the same credentials.
It is in now but with a error message and without any entities.

`Logger: homeassistant.config_entries
Quelle: config_entries.py:604
Erstmals aufgetreten: 21:27:45 (1 Vorkommnisse)
Zuletzt protokolliert: 21:27:45

Error setting up entry STP10.0SE (SUNNY TRIPOWER 10.0 SE) (sw3017493665) for pysmaplus
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/pysmaplus/device_speedwire.py", line 452, in device_list
    await asyncio.wait_for(fut, timeout=self._protocol._overallTimeout)
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 519, in wait_for
    async with timeouts.timeout(timeout):
  File "/usr/local/lib/python3.12/asyncio/timeouts.py", line 115, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 604, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/pysmaplus/__init__.py", line 95, in async_setup_entry
    sma = await getPysmaInstance(hass, entry.data)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/pysmaplus/__init__.py", line 88, in getPysmaInstance
    await sma.new_session()
  File "/usr/local/lib/python3.12/site-packages/pysmaplus/device_speedwire.py", line 432, in new_session
    await self.device_info()
  File "/usr/local/lib/python3.12/site-packages/pysmaplus/device_speedwire.py", line 441, in device_info
    ll = await self.device_list()
         ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pysmaplus/device_speedwire.py", line 463, in device_list
    raise SmaConnectionException("No connection to device")
pysmaplus.exceptions.SmaConnectionException: No connection to device
`

borsti112, 2024-10-09:
After restart it is working now.

littleyoda, 2024-10-09:
great

speedwire or webconnect?

borsti112, 2024-10-09:
Webconnect. 

Actually I have another issue. I wanted to update homeassistant to the current core 2024.10.1. But it cancel the process due to pysma with following message:
Logger: homeassistant.const
Quelle: helpers/deprecation.py:197
Erstmals aufgetreten: 20:56:07 (1 Vorkommnisse)
Zuletzt protokolliert: 20:56:07

POWER_VOLT_AMPERE_REACTIVE was used from pysmaplus, this is a deprecated constant which will be removed in HA Core 2025.9. Use UnitOfReactivePower.VOLT_AMPERE_REACTIVE instead, please report it to the author of the 'pysmaplus' custom integration

Do you know this?

littleyoda, 2024-10-09:
You can safely update.

I'm aware of this. But I still have a year (Sep 2025) to make the change. :-)

closed 2024-10-09 by littleyoda

lucientergal, 2024-10-22:
Be careful, some it device like routers or switchs can be incompatible with SMA speedwire protocol, I experienced that on my network (with swisscom box3 as router). Photovoltaic's installators use a tplink mesh device (i don't have the exact reference in mind, but i can share this detail if necessary) in order to let SHM2 and STP8SE communicate each other. I can retrieve data with modbus TCP HA addon but impossible with speedwire EXCEPTED if i plug my computer on the same TPlink device...