littleyoda/ha-pysmaplus #20: getting warning in log after adding Tripower SE

borsti112, 2024-10-10:
Used Version of Home Assistant: 2024.10.1
Used Version of ha-pysmaplus: 0.4.5.1

**Describe the bug, Remarks, ...**
After adding the Tripower SE with speedwire or Webconnect (both tested) I am getting now following recurrent warnings in the log:

**Used Interface/Access Method**
see https://raw.githubusercontent.com/littleyoda/ha-pysmaplus/main/docs/images/interfaces.png
Speedwire: [ X]
Webconnect: [X ]
EnnexOS-based: [ ]
Energymeter/SHM2: [X ]

**List of the integrated SMA devices**
Inverter:
Energymeter:
Sunny Home Manager 2: X

**Screenshots and Diagnose Information**

`
2024-10-10 11:11:18.296 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00469109: [8395, 0] @ ['62', '66']
2024-10-10 11:11:18.297 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00469209: [135387, 0] @ ['62', '66']
2024-10-10 11:11:18.557 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00464E09: [4, 4, 4, 4, 1] @ ['62', '66', '70', '74', '78']
2024-10-10 11:11:25.303 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00469109: [8395, 0] @ ['62', '66']
2024-10-10 11:11:25.304 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00469209: [135387, 0] @ ['62', '66']
2024-10-10 11:11:25.632 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00464E09: [21, 21, 21, 21, 1] @ ['62', '66', '70', '74', '78']
2024-10-10 11:11:32.283 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00469109: [8395, 0] @ ['62', '66']
2024-10-10 11:11:32.284 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00469209: [135387, 0] @ ['62', '66']
2024-10-10 11:11:32.545 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00464E09: [10, 10, 10, 10, 1] @ ['62', '66', '70', '74', '78']
2024-10-10 11:11:34.900 ERROR (MainThread) [aiohttp.server] Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_protocol.py", line 377, in data_received
    messages, upgraded, tail = self._request_parser.feed_data(data)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "aiohttp/_http_parser.pyx", line 563, in aiohttp._http_parser.HttpParser.feed_data
aiohttp.http_exceptions.BadStatusLine: 400, message:
  Invalid method encountered:

    b''
      ^
2024-10-10 11:11:39.219 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00469109: [8395, 0] @ ['62', '66']
2024-10-10 11:11:39.219 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00469209: [135387, 0] @ ['62', '66']
2024-10-10 11:11:39.501 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00464E09: [2, 2, 2, 2, 1] @ ['62', '66', '70', '74', '78']
2024-10-10 11:11:46.291 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00469109: [8395, 0] @ ['62', '66']
2024-10-10 11:11:46.292 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00469209: [135387, 0] @ ['62', '66']
2024-10-10 11:11:46.575 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00464E09: [3, 3, 3, 3, 1] @ ['62', '66', '70', '74', '78']
2024-10-10 11:11:53.278 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00469109: [8395, 0] @ ['62', '66']
2024-10-10 11:11:53.278 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00469209: [135387, 0] @ ['62', '66']
2024-10-10 11:11:53.546 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00464E09: [8, 8, 8, 8, 1] @ ['62', '66', '70', '74', '78']
2024-10-10 11:11:54.931 ERROR (MainThread) [aiohttp.server] Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_protocol.py", line 377, in data_received
    messages, upgraded, tail = self._request_parser.feed_data(data)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "aiohttp/_http_parser.pyx", line 563, in aiohttp._http_parser.HttpParser.feed_data
aiohttp.http_exceptions.BadStatusLine: 400, message:
  Invalid method encountered:

    b''
      ^
2024-10-10 11:12:00.271 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00469109: [8395, 0] @ ['62', '66']
2024-10-10 11:12:00.272 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00469209: [135388, 0] @ ['62', '66']
2024-10-10 11:12:00.578 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00464E09: [11, 11, 11, 11, 1] @ ['62', '66', '70', '74', '78']
2024-10-10 11:12:07.358 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00469109: [8395, 0] @ ['62', '66']
2024-10-10 11:12:07.358 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00469209: [135388, 0] @ ['62', '66']
2024-10-10 11:12:07.676 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00464E09: [0, 0, 0, 0, 1] @ ['62', '66', '70', '74', '78']
2024-10-10 11:12:14.323 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00469109: [8395, 0] @ ['62', '66']
2024-10-10 11:12:14.324 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00469209: [135388, 0] @ ['62', '66']
2024-10-10 11:12:14.684 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00464E09: [6, 6, 6, 6, 1] @ ['62', '66', '70', '74', '78']
2024-10-10 11:12:14.771 ERROR (MainThread) [aiohttp.server] Error handling request

`

littleyoda, 2024-10-19:
Since there is no official description for Speedwire, all the information had to be reverse engineered. Unfortunately, not all answers are clear yet, so message types that are unknown keep popping up with the "No Handler" notice.

[If I ever have a lot of time on my hands, I'll add something that only displays the messages once.]
The cause of the other error message (aiohttp) is not clear to me. Are you sure that it is triggered by my integration?

snowdd1, 2024-12-08:
I just added an SHM2 and a Tripower 10SE to my new HomeAssitant, and I am seeing the same `No Handler for` messages, plus some more id's. 
I do not get the `aiohttp` errors shown above, so that is probably from some other integration. 
```log
2024-12-08 12:06:25.443 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 08822C09: [16777677, 2340, 8620, 8631, 8632, 8633, 16777213, 16777214] @ ['62', '66', '70', '74', '78', '82', '86', '90']
2024-12-08 12:06:25.509 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00B18909: [0, 0, 0, 0, -1276505928, -1276505928, 0, 0] @ ['62', '66', '70', '74', '78', '82', '86', '90']
2024-12-08 12:06:26.033 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00469109: [2197082, 0] @ ['62', '66']
2024-12-08 12:06:26.034 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00469209: [874395, 0] @ ['62', '66']
2024-12-08 12:06:26.335 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00464E09: [70, 70, 70, 70, 1] @ ['62', '66', '70', '74', '78']
```

How can I help to make the repeated message not being logged? I speak some Python...

snowdd1, 2024-12-08:
I created an MR at https://github.com/littleyoda/pysma/pull/10 - it is still draft because I do not have a testing environment right now.

closed 2025-02-20 by littleyoda