littleyoda/ha-pysmaplus #22: [Feature Request] SMA Data Manager M connected hardware

Wopples, 2024-10-18:
Great work Littleyoda!

The integration managed to connect to our SMA Data Manager M (EDMM-10). It manages 3 Sunny Tripower Core2 invertors, no batteries.

These sensors are live and seem consistent with the Sunny Portal web interface:
Grid Power
Grid Reactive Power
PV Gen Meter
PV Power

These sensors are reporting 0 or "unknown":
Active Power Limitation
Battery Charge Total
Battery Discharge Total
Inverter Power Limit
Power Limit (IO)
Operating Status General (Diagnostic)

Would it be possible to access data from the individual connected inverters? 

If so, I would be happy to help providing documentation or testing.

littleyoda, 2024-10-18:
Did you tested it with the lastet version from yesterday evening? v0.4.6.3?
If not, please test again with the latest version.

Do I understand you correctly?
When adding the Data Manager, you were able create devices for 3 inverters and ‘plant:1’?

If I have understood everything correctly, please send me the [diagnostic files](https://github.com/littleyoda/ha-pysmaplus/blob/main/docs/diagnosticsinformation.md) for each of inverters als a zip file.  I can then check whether the sensors are returned by the Data Manager at all and add necessary sensors.

Wopples, 2024-10-19:
> Did you tested it with the lastet version from yesterday evening? v0.4.6.3? If not, please test again with the latest version.
> 
v0.4.6.3
> Do I understand you correctly? When adding the Data Manager, you were able create devices for 3 inverters and ‘plant:1’?
> 
No. I only have the datamanager as a device via EnnoxOS. The invertors were shown in the selection dialogue, but when selecting them, no device is created.
> If I have understood everything correctly, please send me the [diagnostic files](https://github.com/littleyoda/ha-pysmaplus/blob/main/docs/diagnosticsinformation.md) for each of inverters als a zip [...]
See above. Logging was not active yet. Now it is. I will try to reestablish a link with an inverter on monday and see if a log is created.
> EDIT: What I didn't understand: Did you only connect the three inverters to the data manager or did you also connect other devices, such as battery inverters?
>
We have no batteries. Only the inverters are connected.

littleyoda, 2024-10-19:
ok. Lets start with the basic.

did you get a dialogue like the one below?
were all your devices displayed there?

[image]

Wopples, 2024-10-21:
Hi Sven,
When adding a device the folowing selection appears:
[image]
This is correct. I presume the EDMM-10 is the physical data manager and Luyckx Plant 1 is the same data manager exposed to the Sunny Portal web interface.

Adding an inverter does not create a device:
[image]

Adding the Plant works, but with limited parameters:
[image]

To give you some more context: HA is installed on our company network. We want to use it to do some logging of energy parameters and experiment with peak shaving. Our HA is installed on an Odroid and we have a seperate VLAN for all devices that have network connection and can provide us with data. It 's unclear to me in what way our fire wall and network rules are interfering with the operations of HA.
Over the weekend I noticed that data capture from the plant stoped at the moment I logged out of the network friday evening. Could be a coincidence as the account I use for Sunny Portal has no connection whatsoever with my network account.
[image]

Wopples, 2024-10-21:
Hi Sven,

I suppose you're also interested in loggings.

This has to do with the lack of datapoints between friday evening and monday night. The time of the last logging corresponds more or less with the "reset" to zero (consistent with the lack of solar energy at night):
[image]

I also readded inverter 1 and these were the logs:
[image]
[image]
[image]

littleyoda, 2024-10-22:
Ok. That looks pretty good from my point.

If a device is not yet known (‘Unknown Device 19047’), an empty device is created, exactly as shown in your screenshots.

Can you please send me the  [Diagnostic Information](https://github.com/littleyoda/ha-pysmaplus/blob/main/docs/diagnosticsinformation.md) of one of the two entries?

Wopples, 2024-10-23:
Sven,

Here you go: 
[config_entry-pysmaplus-01JAVZ7GTR2ZSPJBF2N8E632T3 blinded.txt](https://github.com/user-attachments/files/17494275/config_entry-pysmaplus-01JAVZ7GTR2ZSPJBF2N8E632T3.blinded.txt)

company = L*u-y+c&k'x

I also noticed a recurring error every hour loged in the core log:
`2024-10-23 14:37:53.156 ERROR (MainThread) [pysmaplus.device_ennexos] Error requesting https://xx.xx.x.46/api/v1/featuretoggles [Timeout]
2024-10-23 15:37:48.363 ERROR (MainThread) [pysmaplus.device_ennexos] Request to https://xx.xx.x.46/api/v1/measurements/live did not return a valid json. Code 401`

littleyoda, 2024-10-25:
v0.4.6.6 should support the SMA STP 110-60 via Data Manager.

Wopples, 2024-10-26:
Hi Sven,

Updated to v0.4.6.7. The inverter devices are still not created. No error logs in HA Core.
Enclosed the diagnostics:
[config_entry-pysmaplus-01JB282H4CS2FT19AT6WH9P2RS blinded.txt](https://github.com/user-attachments/files/17529224/config_entry-pysmaplus-01JB282H4CS2FT19AT6WH9P2RS.blinded.txt)

littleyoda, 2024-10-26:
Okay. Change of plan.

I would like to try out what the pysma command line tools return in your cases. This should make debugging a lot easier.

Depending on your operating system, download the Linux or Windows version from:
Okay. Change of plan.

I would like to try out what the pysma command line tools return for you.

Depending on your operating system, download the Linux or Windows version from:
https://github.com/littleyoda/pysma/releases/tag/v0.3.18.3

For the Linux version
chmod +x pysma_cli_linux
./pysma_cli_linux -s ennexos username "password" ip

For Windows:#
./pysma_cli_win.exe -s ennexos username "password" ip

A log file (example.log) should be created. I will also be interested in the output of the tool.

Warning: The Windows version occasionally produces a false alarm with virus scanners. I then have to manually request to unlock the version in each case.

littleyoda, 2024-10-26:
ignore my last message for know. I'm pretty sure I've found the error now.

Please try the latest version (0.4.6.8).

Wopples, 2024-10-28:
Sie haben es geschaft!
You did it Sven.
[image]

littleyoda, 2024-10-28:
finally ... ;-)

closed 2024-10-28 by littleyoda