littleyoda/ha-pysmaplus #23: Bug: Can't create integration

mattmeye, 2024-10-22:
Used Version of Home Assistant:
````
Core 2024.10.3
Supervisor 2024.10.2
Operating System 13.2
Frontend 20241002.3
````
Used Version of ha-pysmaplus: 

````
v0.4.6.3
````
**Describe the bug, Remarks, ...**
I get an error during the integration creation: 
````
Der Konfigurationsfluss konnte nicht geladen werden: {"message":"Invalid handler specified"}
````
Steps to reproduce:

- [x] HACS must be installed
- [x] You have to add my repository in HACS. https://github.com/littleyoda/ha-pysmaplus
- [x] Then add the following integration via HACS SMA Devices Plus
- [x] Restart HA
- [ ] Add your SMA devices based on the SMA Device Plus integration. (Settings / Devices & Services / Integrations)

**Used Interface/Access Method (I want to use)**
see https://raw.githubusercontent.com/littleyoda/ha-pysmaplus/main/docs/images/interfaces.png
EnnexOS-based: [x]

**List of the SMA devices (I want to integrate)**
Inverter: SMA Tripower X 15

**Screenshots and Diagnose Information**
If applicable, add screenshots to help explain your problem and send the [Diagnose Information](https://raw.githubusercontent.com/littleyoda/ha-pysmaplus/main/docs/Diagnose-Information.png) to the developer. (E-Mail can be found here: https://littleyoda.github.io/)

littleyoda, 2024-10-23:
fixed in 0.4.6.5

closed 2024-10-23 by littleyoda

mattmeye, 2024-10-23:
Thank you very much! Updated, restarted, works <3