littleyoda/ha-pysmaplus #24: Control sbs5.0

vdvmichel, 2024-11-02:
Is it possible to control sma sbs5.0 
I have dynamic prices and I want that the battery inverter don’t load in the morning when price is high and put the solar energy to the grid…

littleyoda, 2024-11-02:
It depends on the used interface?

Do you use ennexos, speedwire or webconect?
When using, enneoxos oder webconnect, some paramters are supported: 
https://github.com/littleyoda/ha-pysmaplus/blob/main/docs/set_parameter.md

vdvmichel, 2024-11-02:
I think speedwire because my installation is still not viewing in the enneoxos platform…

littleyoda, 2024-11-02:
What options did you choose in home assistant to add the device?
[image]

I checkd my list and the  Sunny Boy Storage should support webconnect.

If only speedwire works for you,  I cannot help you.
Controlling it via Modbus may be possible. As a guide, you will also find a few Modubs addresses on the page linked above. However, since these are not the same for every device, you have to check them in the documentation for your device.

vdvmichel, 2024-11-02:
I have used speedwire

vdvmichel, 2024-11-02:
I will check if webconnect will work

vdvmichel, 2024-11-02:
I just changed it to webconnect and yes it connect…. Is it then possible to set charge en discharge limits?

vdvmichel, 2024-11-02:
This is the documentation I found…
[MODBUS-HTML_SBS3.7-6.0-10-V13.zip](https://github.com/user-attachments/files/17606474/MODBUS-HTML_SBS3.7-6.0-10-V13.zip)

littleyoda, 2024-11-02:
https://github.com/littleyoda/ha-pysmaplus/blob/main/docs/set_parameter.md
[image]

You should be able to use Active Power Limitation or Active Power Limitation GCP.

vdvmichel, 2024-11-02:
Thx but I for me it’s more important to do limitation of charge en discharge so I can do all my solar energy to the grid when the price is high.
I have also huawei Solar and there I do that. Maybe it’s also possible to do that white sma but I don’t find it…
But thanks for your fast reply…

littleyoda, 2024-11-02:
In that case, you first have to find out which parameters are relevant.
It's best to ask in an appropriate forum.
I can't help you here.

vdvmichel, 2024-11-02:
i ask this to chatgpt :-)
 based on the documentation, it seems possible to set the charge and discharge power limits for the battery via the Modbus registers. The following registers are relevant for managing the battery's charge and discharge capacities:

CmpBMS.BatChaMinW – Minimum battery charge capacity

Register address: 40793 or 44431
Type: U32 (unsigned 32-bit integer)
Function: Sets the minimum charge power (in watts).
CmpBMS.BatChaMaxW – Maximum battery charge capacity

Register address: 40795 or 44433
Type: U32
Function: Sets the maximum charge power (in watts).
CmpBMS.BatDschMinW – Minimum battery discharge capacity

Register address: 40797 or 44435
Type: U32
Function: Sets the minimum discharge power (in watts).
CmpBMS.BatDschMaxW – Maximum battery discharge capacity

Register address: 40799 or 44437
Type: U32
Function: Sets the maximum discharge power (in watts).
How to Use These Registers
If the registers are marked as WO (write-only), you should be able to write values to them to set the charge and discharge power limits. To do this in Home Assistant or through a Modbus tool, you’ll need the following configuration:

Modbus Integration in Home Assistant: Add the Modbus configuration in configuration.yaml to access the appropriate registers.

yaml
Code kopiëren
modbus:
  - name: SMA_Battery
    type: tcp
    host: YOUR_SMA_INVERTER_IP
    port: YOUR_MODBUS_PORT
    sensors:
      - name: Max_Charge_Power
        address: 40795
        input_type: holding
        unit_of_measurement: W
        scan_interval: 10
        write_type: holding
        max_value: 10000 # Set to the maximum charge power in W
        min_value: 0 # Set to the minimum charge power in W

      - name: Max_Discharge_Power
        address: 40799
        input_type: holding
        unit_of_measurement: W
        scan_interval: 10
        write_type: holding
        max_value: 10000 # Set to the maximum discharge power in W
        min_value: 0 # Set to the minimum discharge power in W
Python Script for Modbus: You can also set the registers using the pymodbus library. Here’s an example to set the maximum charge power:

python
Code kopiëren
from pymodbus.client.sync import ModbusTcpClient

client = ModbusTcpClient('YOUR_SMA_INVERTER_IP', port=YOUR_MODBUS_PORT)
client.connect()

# Set maximum charge power to 5000 W
client.write_register(40795, 5000, unit=1)

# Set maximum discharge power to 5000 W
client.write_register(40799, 5000, unit=1)

client.close()
Testing and Verifying
Before proceeding, first check if it’s actually possible to write to these registers. Try setting lower values to see if the inverter accepts the settings, and monitor if there’s an immediate change in the battery’s charge and discharge behavior.

If you need help configuring this in Home Assistant or with the Python script, let me know!

littleyoda, 2024-11-02:
As already written, the topic is outside the scope of where I can help.

I would ask the question in a suitable forum. Depending on the language: e.g. (https://www.photovoltaikforum.com/, https://community.home-assistant.io/

You can check if you find these parameters on the local website of the inverter.

vdvmichel, 2024-11-03:
I found something that everyone can use and it will also control my SBS5.0
Forecast-based & Manual Control for Homeassistant from SMA STP SE Hybrid Inverters

https://github.com/Optic00/ha-akkusteuerung-sma

closed 2024-11-03 by littleyoda