littleyoda/ha-pysmaplus #25: Request to https://192.168.1.105/api/v1/token did not return a valid json. Code 400

alexanderbeermann, 2024-11-07:
Used Version of Home Assistant: 2024.10.4
Used Version of ha-pysmaplus: 0.4.6.9

**Describe the bug, Remarks, ...**
First of all, thank you very much for the HACS integration pysmaplus - it is working for almost two month without any issue and gave me easily access to the data of my inverter!

Since yesterday evening I am facing the following error message.
`Logger: pysmaplus.device_ennexos
Quelle: custom_components/pysmaplus/__init__.py:88
Integration: SMA Devices Plus (Dokumentation, Probleme)
Erstmals aufgetreten: 07:00:08 (2 Vorkommnisse)
Zuletzt protokolliert: 07:18:28
Request to https://192.168.1.105/api/v1/token did not return a valid json. Code 400`

and additionally:
`Logger: homeassistant.config_entries
Quelle: config_entries.py:594
Erstmals aufgetreten: 07:00:08 (2 Vorkommnisse)
Zuletzt protokolliert: 07:18:28
Error setting up entry Sunny Tripower X 15 (en3020329164) for pysmaplus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/pysmaplus/__init__.py", line 95, in async_setup_entry
    sma = await getPysmaInstance(hass, entry.data)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/pysmaplus/__init__.py", line 88, in getPysmaInstance
    await sma.new_session()
  File "/usr/local/lib/python3.12/site-packages/pysmaplus/device_ennexos.py", line 160, in new_session
    raise SmaAuthenticationException("Login failed!")
pysmaplus.exceptions.SmaAuthenticationException: Login failed!`

I tried several restarts, but the the issue persists. 
Additionally I reloaded the integration ha-pysmaplus, which still did not solve the issue.

With my user credentials I am able to login into https://ennexos.sunnyportal.com/ - but I am not able to use the same user credentials to login into the web interface of the inverter via the local IP address.
Access to theEnnexos App is still working.

**Used Interface/Access Method**
see https://raw.githubusercontent.com/littleyoda/ha-pysmaplus/main/docs/images/interfaces.png
Speedwire: [ ]
Webconnect: [ ]
EnnexOS-based: [X ]
Energymeter/SHM2: [ ]

**List of the integrated SMA devices**
Inverter: X
Energymeter:
Sunny Home Manager 2:

**Screenshots and Diagnose Information**

Excerpt out of the debugging log:

`2024-11-07 06:41:57.666 INFO (MainThread) [custom_components.pysmaplus] Creating new pysma-Device [Session: True, Verify-SSL: False, Url: https://192.168.1.105, User/Group: ***@****.de, Accessmethod: ennexos]`
 and

```
2024-11-07 06:41:59.360 ERROR (MainThread) [pysmaplus.device_ennexos] Request to https://192.168.1.105/api/v1/token did not return a valid json. Code 400
2024-11-07 06:41:59.361 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry Sunny Tripower X 15 (en3020329164) for pysmaplus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/pysmaplus/__init__.py", line 95, in async_setup_entry
    sma = await getPysmaInstance(hass, entry.data)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/pysmaplus/__init__.py", line 88, in getPysmaInstance
    await sma.new_session()
  File "/usr/local/lib/python3.12/site-packages/pysmaplus/device_ennexos.py", line 160, in new_session
    raise SmaAuthenticationException("Login failed!")
```

alexanderbeermann, 2024-11-07:
To login successfully at https://ennexos.sunnyportal.com/ I had to migrate my SMA-User to an SMA-ID. 
Beside this, something might have changed in the SMA ennexos backend on 5.11.2024 - see the release notes: https://www.sma.de/en/products/monitoring-control/sunny-portal#t05
`Introduction of the "Guest" user right
Users with the guest privilege have read-only access to a system and cannot make any changes
A guest does not have the opportunity to use the SMA Energy App for the corresponding system
`

If there are any data I can provide to analyse this issue further more, I am happy to support.

littleyoda, 2024-11-07:
> With my user credentials I am able to login into https://ennexos.sunnyportal.com/ - but I am not able to use the same user credentials to login into the web interface of the inverter via the local IP address.

Das hört sich nach dem Hauptproblem an. 

Meine Integration braucht die Zugangsdaten für das lokale Webinterface des Wechselrichters.
Zugangsdaten für das sunnyportal spielen keine Rolle.

Solange du dich nicht lokal auf dem Wechselrichter anmelden kannst, wird meine Integration nicht funktionieren.

Ich werde aber nicht schlau daraus,warum es früher bei dir funktioniert hat.

alexanderbeermann, 2024-11-07:
Danke für die schnelle Rückmeldung!
Dann frage ich mal bei meinem Installateur bzw. bei SMA an.
Das Ergebnis kommuniziere ich dann hier.

alexanderbeermann, 2024-11-07:
Hallo, 
mittlerweile funktioniert der Login auf der lokalen IP wieder?!
Ggf wurde da die Inverter Software im Hintergrund aktualisiert. Aktiv angestoßen habe ich nichts.  
Jetzt funktioniert auch der Login durch pysmaplus wieder anstandslos. 
Etwas suspekt das ganze...

closed 2024-11-07 by littleyoda