littleyoda/ha-pysmaplus #27: Unknown Device: 19143 N:EVC22-3AC-20

joshiste, 2024-11-23:
Used Version of Home Assistant: 2024.11.3
Used Version of ha-pysmaplus: v0.4.6.9

Found the following output in the log, therefore I thought to report here.
```
2024-11-23 08:36:38.094 ERROR (MainThread) [pysmaplus.device_ennexos] Unknown Device: 19143 N:EVC22-3AC-20 T:SMA EVC22-3AC-20 3022527147 ID:IGULD:SELF. Please report to the author of pysmaplus.
```

Let me know if this is useful for you, and whether you need additional information. If not feel free to close this issue.

littleyoda, 2024-11-23:
Can you please e-mail the diagnostic information?

https://github.com/littleyoda/ha-pysmaplus/blob/main/docs/diagnosticsinformation.md

joshiste, 2024-11-23:
sent.

littleyoda, 2024-11-23:
Added in v0.4.7.
(v0.4.7 should be available via HACS in 30 minutes)

closed 2024-11-23 by littleyoda