littleyoda/ha-pysmaplus #29: Sunny Home Manager 2.0 stopped working

molabs, 2024-11-27:
Used Version of Home Assistant: 2024.11.3
Used Version of ha-pysmaplus: 0.4.7

Description:
I had my sunny home manager 2.0 integrated for a while. A few days ago, the communication stopped working. I then wanted to redo the integration, but now I always get the following message when I want to add the sunny home manager:
Failed to connect. Please check hostname/ip.
Since I don't have to enter an IP or anything similar for the sunny home manager integration, I don't understand why it suddenly doesn't work anymore.

**Used Interface/Access Method**
see https://raw.githubusercontent.com/littleyoda/ha-pysmaplus/main/docs/images/interfaces.png
Speedwire: [ ]
Webconnect: [ ]
EnnexOS-based: [ ]
Energymeter/SHM2: [X]

**List of the integrated SMA devices**
Inverter: Sunny Tripower 6.0
Sunny Home Manager 2: 1

littleyoda, 2024-11-28:
Can you check whether there are any entries in the log file?

molabs, 2024-11-28:
I think Home Assistant is trying to fool me. I just wanted to provoke the error again to create a current log entry, and the integration worked straight away. Yesterday evening it didn't work 5-6 times.

littleyoda, 2024-11-28:
If the error occurs again, please check directly whether you can find anything in the log file relating to pysma.

closed 2024-11-28 by littleyoda