littleyoda/ha-pysmaplus #30: Integration fails to load from v0.4.4 on, v0.4.3 is fine: Platform pysmaplus.config_flow not found

tanuva, 2024-12-04:
Used Version of Home Assistant: 2024.11.3
Used Version of ha-pysmaplus: v0.4.4

**Describe the bug, Remarks, ...**
I recently upgraded from HA 2024.08.x to 2024.11.3 and pulled up pysmaplus up in the same run. After the upgrade, HA refused to load the extension:
```
Logger: homeassistant.config_entries
Quelle: config_entries.py:604
Erstmals aufgetreten: 10:38:57 (1 Vorkommnisse)
Zuletzt protokolliert: 10:38:57

Error importing platform config_flow from integration pysmaplus to set up pysmaplus configuration entry: Platform pysmaplus.config_flow not found
```

Downgrading ha-pysmaplus again reveals that v0.4.3 is still fine while v0.4.4 fails to load.

This is HA in Docker, but no HACS. I have ha-pysmaplus checked out and symlinked `ha-pysmaplus/custom_components/pysmaplus` into `.homeassistant/custom_components`.

**Used Interface/Access Method**
see https://raw.githubusercontent.com/littleyoda/ha-pysmaplus/main/docs/images/interfaces.png
Speedwire: [ ]
Webconnect: [x]
EnnexOS-based: [ ]
Energymeter/SHM2: [ ]
SHM2 with Grid Guard Code: [ ]

**List of the integrated SMA devices**
Inverter: SBSE 4.0
Energymeter:
Sunny Home Manager 2: yes

**Screenshots and Diagnose Information**
If applicable, add screenshots to help explain your problem and send the [Diagnose Information](https://raw.githubusercontent.com/littleyoda/ha-pysmaplus/main/docs/Diagnose-Information.png) to the developer. (E-Mail can be found here: https://littleyoda.github.io/)

littleyoda, 2024-12-04:
Do I understand you correctly?
You have tried the latest version (0.4.7). This already produced the error. Testing of other versions showed that 0.4.4 also gives the error, but 0.4.3 works?

Can you please check whether you have any other error messages regarding pysma in the log files.

You are the first to report this error, although countless users are using 2024.11.x.
You could at most try whether it works if you copy the data and don't just use a symlink.

By the way:
May I ask why you don't use hacs?

tanuva, 2024-12-04:
> You have tried the latest version (0.4.7). This already produced the error. Testing of other versions showed that 0.4.4 also gives the error, but 0.4.3 works?

Exactly.

> Can you please check whether you have any other error messages regarding pysma in the log files.

Yes. Alas, this is the only line that HA logs about the extension, in raw output this time:
```
2024-12-04 14:36:28.253 ERROR (MainThread) [homeassistant.config_entries] Error importing platform config_flow from integration pysmaplus to set up pysmaplus configuration entry: Platform pysmaplus.config_flow not found
```

> You are the first to report this error, although countless users are using 2024.11.x.

I feared that response. If more people had run into this it would already have been fixed. It sounds like HA might have something cached or so, but the files in `__pycache__` are up to date (and I deleted the directory manually for good measure). Do you have any ideas if I can add logging somewhere to debug this?

> You could at most try whether it works if you copy the data and don't just use a symlink.

The behavior stays the same. (Also for 2 other extensions I installed this way. The only downside compared to HACS is no auto-updates.)

> May I ask why you don't use hacs?

I *think* my HA installation already existed before there was HA Supervised and I always had more things running on the Pi, so it was just one of more services. (IOW: I wanted to stay in control of the OS.) I just recently switched to Docker because I was tired of compiling bleeding edge Python for Debian stable to run HA on bare metal. :)

After writing most of my reply and doing more reseach: Wait, HACS also works with a plain HA Docker container? I might look into that.

tanuva, 2024-12-19:
I would really love to understand why it didn't work correctly with my checkout. I just tried installing pysmaplus via HACS and... 0.4.7 works fine. Well. I'll accept that for now and tend to other things... :)

closed 2024-12-19 by tanuva