littleyoda/ha-pysmaplus #32: host SMA Solar device

menvis, 2024-12-31:
I have access to the dashboard of "Zonneweide Glimmen" via my internet browser using the "SMA SUNNY PORTAL powered by ennexOS". I want to use the data via the HACS integration SMA Devices Plus in the Home Assistant. Then I select "SMA devices with EnnexOS". The integration then asks for "Host". What do I then have to fill in? Or does this integration not work in my case? See the screen shots.
[image]
[image]
[image]

littleyoda, 2024-12-31:
The Sunny Portal is not supported.
Devices that use the ennexos operating system are supported.

Are there any reasons why you are not adding the individual devices?

menvis, 2024-12-31:
Thank you. I will try the individual devices.

closed 2024-12-31 by littleyoda