littleyoda/ha-pysmaplus #33: SI6.0 seems to constantly lose its connection via Speedwire

medicus07, 2025-01-17:
I use 4 SMA devices (STP25, STP10, SI6 and HM2).
All were recognized as Speedwire devices via the discovery and were also configured as such.
The SI6 is a device from 2018 with an older firmware that I do not want to update (complete reconfiguration).
The entities of the SI6 switch back and forth between “not available” and normal status approx. every 20-30 seconds.

Home Assistant:
Core
2025.1.2
Supervisor
2024.12.3
Operating System
14.1

ha-pysmaplus:
0.4.7

[image]

[image]

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 507, in wait_for
    return await fut
           ^^^^^^^^^
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/custom_components/pysmaplus/__init__.py", line 129, in async_update_data
    await sma.read(sensor_def, entry.data[CONF_DEVICE])
  File "/usr/local/lib/python3.13/site-packages/pysmaplus/device_speedwire.py", line 514, in read
    raise e
  File "/usr/local/lib/python3.13/site-packages/pysmaplus/device_speedwire.py", line 509, in read
    await asyncio.wait_for(fut, timeout=self._protocol._overallTimeout)
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 506, in wait_for
    async with timeouts.timeout(timeout):
               ~~~~~~~~~~~~~~~~^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/timeouts.py", line 116, in __aexit__
    raise TimeoutError from exc_val
TimeoutError
2025-01-17 21:56:27.849 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00462607: [19314223, 0] @ ['62', '66']
2025-01-17 21:56:27.849 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00462707: [113, 0] @ ['62', '66']
2025-01-17 21:56:27.849 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00462807: [6735, 0] @ ['62', '66']
2025-01-17 21:56:30.508 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00469101: [110380021, 0] @ ['62', '66']
2025-01-17 21:56:30.508 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00469201: [5427482, 0] @ ['62', '66']
2025-01-17 21:56:30.781 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00464E01: [70, 70, 70, 70, 1] @ ['62', '66', '70', '74', '78']
2025-01-17 21:56:35.511 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00493307: [5550, 5550, 5550, 5550, 1] @ ['62', '66', '70', '74', '78']
2025-01-17 21:56:36.504 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00469101: [110380021, 0] @ ['62', '66']
2025-01-17 21:56:36.504 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00469201: [5427483, 0] @ ['62', '66']
2025-01-17 21:56:36.780 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00464E01: [69, 69, 69, 69, 1] @ ['62', '66', '70', '74', '78']
2025-01-17 21:56:36.879 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00462607: [19314225, 0] @ ['62', '66']
2025-01-17 21:56:36.879 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00462707: [113, 0] @ ['62', '66']
2025-01-17 21:56:36.879 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00462807: [6737, 0] @ ['62', '66']
2025-01-17 21:56:37.683 WARNING (MainThread) [custom_components.pysmaplus] Update Failed <class 'TimeoutError'> 192.168.1.30/speedwireinv/3006201251
2025-01-17 21:56:37.683 WARNING (MainThread) [custom_components.pysmaplus] 
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 507, in wait_for
    return await fut
           ^^^^^^^^^
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/custom_components/pysmaplus/__init__.py", line 129, in async_update_data
    await sma.read(sensor_def, entry.data[CONF_DEVICE])
  File "/usr/local/lib/python3.13/site-packages/pysmaplus/device_speedwire.py", line 514, in read
    raise e
[... 249 lines cut ...]
    async with timeouts.timeout(timeout):
               ~~~~~~~~~~~~~~~~^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/timeouts.py", line 116, in __aexit__
    raise TimeoutError from exc_val
TimeoutError
2025-01-17 21:59:16.686 ERROR (MainThread) [custom_components.pysmaplus] Error fetching pysmaplus data: 
2025-01-17 21:59:17.201 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00462607: [19314260, 0] @ ['62', '66']
2025-01-17 21:59:17.202 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00462707: [113, 0] @ ['62', '66']
2025-01-17 21:59:17.202 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00462807: [6772, 0] @ ['62', '66']
2025-01-17 21:59:18.530 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00469101: [110380021, 0] @ ['62', '66']
2025-01-17 21:59:18.531 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00469201: [5427518, 0] @ ['62', '66']
2025-01-17 21:59:18.802 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00464E01: [70, 70, 70, 70, 1] @ ['62', '66', '70', '74', '78']
2025-01-17 21:59:24.041 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00493307: [5550, 5550, 5550, 5550, 1] @ ['62', '66', '70', '74', '78']
2025-01-17 21:59:24.554 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00469101: [110380021, 0] @ ['62', '66']
2025-01-17 21:59:24.554 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00469201: [5427520, 0] @ ['62', '66']
2025-01-17 21:59:24.823 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00464E01: [70, 70, 70, 70, 1] @ ['62', '66', '70', '74', '78']
2025-01-17 21:59:24.866 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00462607: [19314261, 0] @ ['62', '66']
2025-01-17 21:59:24.867 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00462707: [113, 0] @ ['62', '66']
2025-01-17 21:59:24.867 WARNING (MainThread) [pysmaplus.device_speedwire] No Handler for 00462807: [6773, 0] @ ['62', '66']
2025-01-17 21:59:25.567 INFO (MainThread) [custom_components.pysmaplus] Fetching pysmaplus data recovered
2025-01-17 21:59:26.441 INFO (MainThread) [custom_components.pysmaplus] Update pysma 192.168.1.31/speedwireinv/1901348931
2025-01-17 21:59:28.394 INFO (MainThread) [custom_components.pysmaplus] Update pysma localhost/speedwireem/3004905827
2025-01-17 21:59:28.436 DEBUG (MainThread) [custom_components.pysmaplus] Finished fetching pysmaplus data in 1.996 seconds (success: True)

```

medicus07, 2025-01-21:
After restarting HA, the device can no longer be set up...

```
Logger: homeassistant.config_entries
Quelle: config_entries.py:640
Erstmals aufgetreten: 19:33:18 (2 Vorkommnisse)
Zuletzt protokolliert: 19:34:00

Error setting up entry SI 6.0H-12 (Sunny Island 6.0H) (sw3006201251) for pysmaplus
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 507, in wait_for
    return await fut
           ^^^^^^^^^
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/pysmaplus/__init__.py", line 103, in async_setup_entry
    sensor_def = await sma.get_sensors(entry.data[CONF_DEVICE])
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pysmaplus/device_speedwire.py", line 497, in get_sensors
    raise e
  File "/usr/local/lib/python3.13/site-packages/pysmaplus/device_speedwire.py", line 492, in get_sensors
    await asyncio.wait_for(fut, timeout=self._protocol._overallTimeout)
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 506, in wait_for
    async with timeouts.timeout(timeout):
               ~~~~~~~~~~~~~~~~^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/timeouts.py", line 116, in __aexit__
    raise TimeoutError from exc_val
TimeoutError
```

and

```
Logger: homeassistant.config_entries
Quelle: config_entries.py:640
Erstmals aufgetreten: 19:33:40 (2 Vorkommnisse)
Zuletzt protokolliert: 19:33:49

Error setting up entry SI 6.0H-12 (Sunny Island 6.0H) (sw3006201251) for pysmaplus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/pysmaplus/__init__.py", line 95, in async_setup_entry
    sma = await getPysmaInstance(hass, entry.data)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/pysmaplus/__init__.py", line 88, in getPysmaInstance
    await sma.new_session()
  File "/usr/local/lib/python3.13/site-packages/pysmaplus/device_speedwire.py", line 434, in new_session
    raise SmaConnectionException(
        f"No connection to device: {self._host}:9522  ({self._protocol._failedCounter}/{self._protocol._sendCounter})"
    )
pysmaplus.exceptions.SmaConnectionException: No connection to device: 192.168.1.30:9522  (3/3)

```

medicus07, 2025-01-22:
`Logger: pysmaplus.device_speedwire
Quelle: runner.py:154
Erstmals aufgetreten: 21. Januar 2025 um 21:36:13 **### (19853 Vorkommnisse)**
Zuletzt protokolliert: 08:38:23

No Handler for 00469201: [5445576, 0] @ ['62', '66']
No Handler for 00464E01: [41, 41, 41, 41, 1] @ ['62', '66', '70', '74', '78']
No Handler for 00469101: [110431377, 0] @ ['62', '66']
No Handler for 00464E01: [42, 42, 42, 42, 1] @ ['62', '66', '70', '74', '78']
No Handler for 00469201: [5445577, 0] @ ['62', '66']
`

littleyoda, 2025-01-22:
1.) Speedwire funktioniert leider nicht in allen Fällen und ich habe bislang den Grund noch  nicht gefunden:
siehe https://github.com/littleyoda/ha-pysmaplus/issues/18

Ich warte noch auf jemanden, der folgende Integration ausprobiert und mir berichtet, ob diese Integration auch das Problem hat oder ob es in diesem Fall ohne Probleme geht:
https://github.com/Wired-Square/homeassistant-sma-sw

2.)  siehe https://github.com/littleyoda/ha-pysmaplus/issues/20
Ab einer der nächsten Vesionen werden die Hinweise nur noch einmal am Tag(oder so) kommen.

medicus07, 2025-01-23:
Versuche spätestens am Wochenende die Integration [homeassistant-sma-sw](https://github.com/Wired-Square/homeassistant-sma-sw) zu testen... melde mich. Danke.

michaelkrause-SAP, 2025-01-23:
Ich nutze EVCC als Addon zu HA und hole mir die summierten Werte über die Mqtt schnittstellle und bin mega zufrieden: https://docs.evcc.io/docs/reference/configuration/mqtt

medicus07, 2025-01-23:
Kenne EVCC, ist für meine Zwecke nicht brauchbar, weil mir die min polling time mit 30 sec zu lang ist. Trotzdem danke!

medicus07, 2025-01-23:
> Versuche spätestens am Wochenende die Integration [homeassistant-sma-sw](https://github.com/Wired-Square/homeassistant-sma-sw) zu testen... melde mich. Danke.

Integration habe ich installiert, Devices funktionieren aber nicht, Entitäten sind nicht verfügbar. 

`Dependencies

This integration relies on the library at https://github.com/Wired-Square/sma-query

The sma-query library should first be installed into Home Assistants virtual environment.`

Hab keine Ahnung, wie ich das machen soll...

littleyoda, 2025-01-23:
ok. den Punkt habe ich überlesen. Schade, doch nicht so einfach zu testen. 

Leider kann ich dir aktuell nicht helfen.

littleyoda, 2025-01-23:
Merke gerade, dass ich den falschen Link eingefügt habe. Sorry.

https://github.com/eddso/ha_sma_speedwire

michaelkrause-SAP, 2025-01-24:
@medicus07 : bei mir steht das interval auf 10s # control cycle interval, läuft sehr stabil. 
Ich hatte es (als es noch auf separater Hardware war) schon auf 5 Sekunden ... ging auch gut.
Der Riesen Vorteil ist halt, dass Du Dir die Werte nicht selbst zusammenrechnen musst ... aber wenn du kein EVCC brauchst, dann wäre es wohl auch "oversized". .. lg Michael

ghe6hs3, 2025-01-26:
Hallo,
ggf. hat uns SMA hier einen Bärendienst erwiesen. Vor einigen Tagen stellte meine VM, die mir die Daten vom Home Manager einsammelt und in den MQTT Broker schiebt, mit dem gleichen Verhalten den Dienst eingestellt. Ich konnte feststellen, dass die Last anstieg und keine Daten abgerufen wurden. 
Da ich gerade den Umstieg von openHAB auf HA evaluiere, habe ich geschaut ob eine Umsetzung mit HA besser funktioniert - bin aber auf das gleiche Verhalten gestoßen und dann hier gelandet. Da ich im Debugging in HA noch nicht ganz fit bin, habe ich mir die Kommunikation angeschaut. Mir ist im Log aufgefallen, dass SMA mir vor einigen Tagen ein Update in den Home Manager eingespielt hat. Die Verbindung scheint jetzt nach einiger Zeit geschlossen zu werden. Ich vermute, dass die Implementierung meiner SMAtoMQTT-Lösung und ggf. auch dieses Addon von einer offenen Verbindung ausgehen. 

Ich habe mir ein Python oder Bash Script gesucht, um die Verbindung zu testen bzw. meine Vermutung zu bestätigen. https://github.com/datenschuft/SMA-EM stellt ein einfaches Python Script bereit, was den Zweck erfüllt. Ich habe [sma-em-measurement.py](https://github.com/datenschuft/SMA-EM/blob/master/sma-em-measurement.py) so umgeschrieben, dass es nicht in einer Endlosschleife läuft - mir reicht der einmalige Abruf um die Verbindung zu öffnen. Damit ließ sich das Problem in HA direkt beheben, alle Daten wurden sofort wieder ausgelesen und angezeigt. Ein Cronjob alle 2 min, der das Script mit dem einmaligen Abruf ausführt, sorgt jetzt dafür dass HA permanent seine Daten erhält.

Klar - das ist keine Lösung. Aber es zeigt vielleicht, an welcher Stelle das Problem schlummern könnte.

littleyoda, 2025-01-27:
@ghe6hs3 
Das Problem hat aber nichts mit dem oben beschriebenen Problem zu tun.
Wenn ich dich richtig verstanden habe, hast du Probleme mit dem Sunny Home Manager 2 oder einem Energymeter.
Auf diese wird ganz anders zugegriffen, als auf die Wechselrichter.

=> https://github.com/littleyoda/pysma/blob/master/doc/faq_DE.md#die-verschiedenen-zugriffsmethoden

CSolero, 2025-01-29:
Hallo!

Habe leider auch das Problem, dass ständig die Verbindung verloren geht:

> Meldung 1:
> 
> Logger: custom_components.pysmaplus
> Quelle: helpers/update_coordinator.py:411
> Integration: SMA Devices Plus (Dokumentation, Probleme)
> Erstmals aufgetreten: 28. Januar 2025 um 09:15:26 (719 Vorkommnisse)
> Zuletzt protokolliert: 08:46:00
> 
> Error fetching pysmaplus data:
> 
> 
> Meldung 2:
> 
> Logger: custom_components.pysmaplus
> Quelle: custom_components/pysmaplus/__init__.py:135
> Integration: SMA Devices Plus (Dokumentation, Probleme)
> Erstmals aufgetreten: 28. Januar 2025 um 09:15:26 (884 Vorkommnisse)
> Zuletzt protokolliert: 08:46:10
> 
> Update Failed <class 'TimeoutError'> 192.168.5.42/speedwireinv/304600738
> 
> 
> Meldung3:
> 
> Logger: custom_components.pysmaplus
> Quelle: custom_components/pysmaplus/__init__.py:129
> Integration: SMA Devices Plus (Dokumentation, Probleme)
> Erstmals aufgetreten: 28. Januar 2025 um 09:15:26 (884 Vorkommnisse)
> Zuletzt protokolliert: 08:46:10
> 
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.13/asyncio/tasks.py", line 507, in wait_for
>     return await fut
>            ^^^^^^^^^
> asyncio.exceptions.CancelledError
> 
> The above exception was the direct cause of the following exception:
> 
> Traceback (most recent call last):
>   File "/config/custom_components/pysmaplus/__init__.py", line 129, in async_update_data
>     await sma.read(sensor_def, entry.data[CONF_DEVICE])
>   File "/usr/local/lib/python3.13/site-packages/pysmaplus/device_speedwire.py", line 514, in read
>     raise e
>   File "/usr/local/lib/python3.13/site-packages/pysmaplus/device_speedwire.py", line 509, in read
>     await asyncio.wait_for(fut, timeout=self._protocol._overallTimeout)
>   File "/usr/local/lib/python3.13/asyncio/tasks.py", line 506, in wait_for
>     async with timeouts.timeout(timeout):
>                ~~~~~~~~~~~~~~~~^^^^^^^^^
>   File "/usr/local/lib/python3.13/asyncio/timeouts.py", line 116, in __aexit__
>     raise TimeoutError from exc_val
> TimeoutError

Außerdem habe ich das Problem, dass nach einem Neustart keine Verbindung zustande kommt und ich jedesmal die Integration neu laden muss, damit sie funktioniert!

> Logger: homeassistant.config_entries
> Quelle: config_entries.py:640
> Erstmals aufgetreten: 28. Januar 2025 um 09:08:40 (1 Vorkommnisse)
> Zuletzt protokolliert: 28. Januar 2025 um 09:08:40
> 
> Error setting up entry STP 5000TL-20 (sw304600738) for pysmaplus
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.13/asyncio/tasks.py", line 507, in wait_for
>     return await fut
>            ^^^^^^^^^
> asyncio.exceptions.CancelledError
> 
> The above exception was the direct cause of the following exception:
> 
> Traceback (most recent call last):
>   File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
>     result = await component.async_setup_entry(hass, self)
>              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/config/custom_components/pysmaplus/__init__.py", line 103, in async_setup_entry
>     sensor_def = await sma.get_sensors(entry.data[CONF_DEVICE])
>                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/usr/local/lib/python3.13/site-packages/pysmaplus/device_speedwire.py", line 497, in get_sensors
>     raise e
>   File "/usr/local/lib/python3.13/site-packages/pysmaplus/device_speedwire.py", line 492, in get_sensors
>     await asyncio.wait_for(fut, timeout=self._protocol._overallTimeout)
>   File "/usr/local/lib/python3.13/asyncio/tasks.py", line 506, in wait_for
>     async with timeouts.timeout(timeout):
>                ~~~~~~~~~~~~~~~~^^^^^^^^^
>   File "/usr/local/lib/python3.13/asyncio/timeouts.py", line 116, in __aexit__
>     raise TimeoutError from exc_val
> TimeoutError

Vielleicht kann jemand damit was anfangen und mir Helfen.

Auf jeden Fall mal Danke im Vorraus.

Lg

NightlyStar, 2025-02-02:
also diese ha_sma_speedwire-Implementierung läuft da besser. 
Kein Ausfall 
Ich besitze ein STP20000TL-30 und nutze da auch Speedwire

littleyoda, 2025-02-02:
Danke für die Rückmeldung. Gibst du in einer Woche oder so nochmal Rückmeldung, ob es immer noch problemlos läuft?
Ich würde mir dann mal die Unterschiede anschauen.

DeltaTango69, 2025-02-06:
Hallo Leute,
ich habe auch ständig Verbundungsprobleme. Den Home Manager 2 konnte ich sogar nur finden indem ich kurz den Sunny-Explorer gestartet habe. Danach wurde er in Homeassistant gefunden und lief ein paar Minuten. Dann war er wieder weg. Sobal ich den Sunny-Explorer starte und der Versucht eine Verbindung aufzubauen geht es wieder. Vielleicht könnt ihr das auch mal testen.

> Ich habe mir ein Python oder Bash Script gesucht, um die Verbindung zu testen bzw. meine Vermutung zu bestätigen. https://github.com/datenschuft/SMA-EM stellt ein einfaches Python Script bereit, was [...]

Werde das mal testen - das deutet ja auf die gleiche Lösung hin.....

DeltaTango69, 2025-02-08:
Bekomme cih leider nicht zum laufen?!

```
dirk@Homeserver:/opt/smaemd$ /opt/smaemd/sma-em-measurement.py

Traceback (most recent call last):
  File "/opt/smaemd/sma-em-measurement.py", line 82, in <module>
    print ('SMA-EM Serial:{}'.format(emparts['serial']))
                                     ~~~~~~~^^^^^^^^^^
KeyError: 'serial'
```

DeltaTango69, 2025-02-08:
So ich habe jetzt auf Unicast umgeschaltet - es läuft jetzt schon eine Weile stabil.... ich schätze es liegt an meiner Nezwerkkonfiguration. 1 Wechselrichter in der Garage ist mit Powerline angeschlossen. der zweite, der Homemanager und der Homeseerver hängt an einem ungemanagden Switch der direkt an die Fritzbox geht. Der weite Wechselrichter machte nie Probleme, nur der erste machte Probleme und der Homemanager läuft gar nicht im Multicast Betrieb. Hoffen wir mal das es jetzt so endlich läuft....

littleyoda, 2025-02-09:
@NightlyStar 
Wie sieht es aus? Immer noch keine Probleme?

Wenn ja, werde ich mal ganz genau die verschickten Pakete vergleichen müssen.
Dafür werde ich aber in den nächsten Woche sicherlich nicht die Resourcen haben.

NightlyStar, 2025-02-09:
Hi
die ha_sma_speedwire Implementation hat bisher bei mir kein Ausfall. Man kann leider sich nur 3. Werte anzeigen lassen. 

Intresannt bei mir ist das bei pysmaplus das Daily Yield und das Total Yield aber auch sauber laufen nur die einzelnen Werte buggen  mit der Fehlermeldung "Diese Entität wird nicht mehr von der „pysmaplus"-Integration bereitgestellt." rum. 

Trozdem gute Arbeit :-) Gruß aus Hessen

lanwin, 2025-02-27:
@littleyoda leider ist das problem mit den Timeouts immer noch da. Ich hatte mir gerade den Code von ha_sma_speedwire angesehe. Dabei ist mir aufgefallen das dort bei jedem cmd nen retry vorgesehen ist. Evtl ist der ground warum die extension funktioniert. https://github.com/eddso/ha_sma_speedwire/blob/07b2f222b9a6d378067a6768b440adee522d8f72/custom_components/sma_speedwire/sma_speedwire.py#L243

lanwin, 2025-03-25:
@littleyoda warscheinlich hast du momentan nicht viel Zeit. Ich denke inzwischen das es tatsächlich hauptsächlich die Retry Logic ist weswegen ha_sma_speedwire zuverlässiger ist. Hab beide parallel laufen und wenn ha_sma_speedwire noch Werte anzeigt und ha-pysmaplus keine mehr, dann tut es 1 - 2 reloads von ha-pysmaplus. Wenn die Lib von sich aus 3 versuche mit wachsendem Timeout versuchen würde, müsste es vermutlich zuverlässiger klappen.

littleyoda, 2025-03-25:
[Ja, bin gerade zeitlich/gesundheitlich eingeschränkt]

Es werden bereits mehrere Retries-Versuche durchgeführt. 
Du kannst mal die Speedwire Integration neu hinzufügen und dann kannst du in der aktuellen Version die retry-Zahl selber festlegen. Kannst es ja mal probieren, wenn du retry auf 5 oder so setzt.

lanwin, 2025-03-25:
Hmm die Anzahl der Retries ist es leider nicht. Mit 5 sehe ich keinen Unterschied.

[image]

Momentan kann man sehen das jeder er immer wieder Aussetzer hat. 

Ich versuche jetzt mal was passiert wenn ich das Updateinterval senke. Gibt es ne Möglichkeit längere Timeouts zu testen?

lanwin, 2025-11-09:
@littleyoda ich habs gerade mal umgestellt. Grundsätzlich läuft es. Ist es Absicht das es damit nur noch Yield und Daily Yield gibt und die anderen Werte nicht mehr?

Darf ich Fragen was du da intern geändert hast?

lanwin, 2025-11-10:
> Es ist eine komplette neu Implementierung, da ich die Ursprache für die Speedwire-Probleme in der alten Version nicht finden konnte und ich testen will, ob eine neue Implementierung besser funktioniert.

Also Entwickler hätte mich mal interessiert wo der Unterschied bei der technischen Herrangehensweise ist.

lanwin, 2025-11-13:
Aber die Sensoren sind ja da. Das was in der anderen Issue steht sollte ja nur zutreffen wenn HA nachts neu gestartet wird. Ich hab aber nur die Integration neu geladen. Die Sensoren sind alle noch da. Werden aber nicht geupdated.

BTW kannts du nicht einfach die Sensoren regelmäßig in eine Datei schreiben (pro IP) und die dann beim starten laden. Dann müsste man nur einmal am Tag die integration neu starten damit alle sensoren da sind und bei den nächsten neustarts von HA wären die dann immer da.

lanwin, 2025-11-17:
@DeltaTango69 @CSolero habt ihr die integration gelöscht und die V2 neu eingefügt. Ich hab damit tatsächlich weniger probleme.

DeltaTango69, 2025-11-17:
Die sind dann weg, weil die Entitäten sehr wahrscheinlich andere Namen bekommen. Das Risiko war es mir bisher nicht Wert.

lanwin, 2026-01-30:
@littleyoda V2 läuft bei mir seit ner Weile jetzt sehr gut. Ich hab ab und mal noch einmal nen einzelnen Datenpunkt der Unavilable ist. Aber sehr viel weniger als vorher.