littleyoda/ha-pysmaplus #34: Mapping sensors to home assistant dashboard

nabbl, 2025-01-18:
Hi there,

I have trouble mapping the sensors from my TriPower 8.0 to the home assistant energy dashboard:

I need the following values:

Grid Feed
Amount of KwH needed from Grid
PV Generation

So Grid Feed I probably need to calculate myself?

These are the sensors available:

[image]

How would I map these correctly?

littleyoda, 2025-01-18:
These two faq-pages should actually help you.:

https://github.com/littleyoda/ha-pysmaplus/blob/main/docs/dashboard.md
https://github.com/littleyoda/ha-pysmaplus/blob/main/docs/faq.md#current-consumption

nabbl, 2025-01-18:
Exactly what I need! Thy

closed 2025-01-18 by littleyoda