littleyoda/ha-pysmaplus #35: Using of multiple SHM2/EM devices?

maikt, 2025-01-18:
Used Version of Home Assistant: 2025.1.2
Used Version of ha-pysmaplus: 0.4.7

Hello,
at my home I use an SMA home manager 2.0 (for the home) and an SMA Entergy Meter 20 for an separate PV system. The EM sends the data with a different multicast address (239.12.255.253) and so I can get the data of both devices in openhab.

Now I check the switch from openhab to HA and I have tested the SMA Devices Plus integration. So I have installed the integration via HACS and try to add the devices via Setup "SMA Energy Meter / Sunny Home Manager 2.0". My problem is, only the SHM2 will be found and integrated, but there is no chance to find the EM20.

Am I doing something wrong? Or if it possible to query the multicast address before the integration starts searching for the SHM2/EM2 devices?

Thanks

littleyoda, 2025-01-18:
Interesting problem:

My integration only listens on 239.12.255.254:9522. There is no need to use different addresses, becausethe integration ist able to distinguish between different EnergyMeters/SHM2 based on the serial number..

How did you change the multicast address? I wasn't aware that this is possible.

maikt, 2025-01-18:
Yes it is possible to change the multicast address on the EnergyMeter settings:

[image]

And I have to change the multicast address because both devices (EM20 and SHM2) work on different power connections (electricity meters) but in the same Ethernet network.

littleyoda, 2025-01-18:
good to know.

Actually, there both devices should work using the same multicast address. So give this a try.

maikt, 2025-01-18:
Great, it works, Thanks a million 🙂 

In contrast to your great integration, the openhab sma binding could not distinguish the devices based on the serial number and so I must distinguish the devices with the multicast address. But that is no longer necessary.

closed 2025-01-18 by littleyoda