littleyoda/ha-pysmaplus #36: Device unavailable after update to 0.4.8

lanwin, 2025-02-05:
Used Version of Home Assistant: 2025.1.4
Used Version of ha-pysmaplus: 0.4.8

**Describe the bug, Remarks, ...**
Device is unavailable in HW after update to 0.4.8. After downgrade to 0.4.7 it works again.

**Used Interface/Access Method**
see https://raw.githubusercontent.com/littleyoda/ha-pysmaplus/main/docs/images/interfaces.png
Speedwire: [x]
Webconnect: [ ]
EnnexOS-based: [ ]
Energymeter/SHM2: [ ]
SHM2 with Grid Guard Code: [ ]

**List of the integrated SMA devices**
Inverter: STP 10000TL-20

**Screenshots and Diagnose Information**

[image]

In HA Log I see the following error.

```
2025-02-05 08:52:02.826 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry STP 10000TL-20 (sw305048063) for pysmaplus
homeassit  | Traceback (most recent call last):
homeassit  |   File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
homeassit  |     result = await component.async_setup_entry(hass, self)
homeassit  |              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
homeassit  |   File "/config/custom_components/pysmaplus/__init__.py", line 109, in async_setup_entry
homeassit  |     sma_device_info = device_list[entry.data[CONF_DEVICE]]
homeassit  |                       ~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^
homeassit  | KeyError: '305048063'
```

littleyoda, 2025-02-08:
Was passiert, wenn du die Integration neu lädst?

Es gibt immer noch ein Problem mit Speedwire, dass Geräte nicht sofort erkannt werden.

https://github.com/littleyoda/ha-pysmaplus/blob/main/docs/faq.md#speedwire-inverter-and-battery-storage

lanwin, 2025-02-08:
Hmm ich wollte gerade schreiben das beim neuladen der gleiche Fehler passiert. Aber auch das Downgrade auf 4.7. zeigt aktuell den Fehler. 

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 507, in wait_for
    return await fut
           ^^^^^^^^^
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 637, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/pysmaplus/__init__.py", line 111, in async_setup_entry
    sensor_def = await sma.get_sensors(entry.data[CONF_DEVICE])
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pysmaplus/device_speedwire.py", line 529, in get_sensors
    raise e
  File "/usr/local/lib/python3.13/site-packages/pysmaplus/device_speedwire.py", line 524, in get_sensors
    await asyncio.wait_for(fut, timeout=self._protocol._overallTimeout)
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 506, in wait_for
    async with timeouts.timeout(timeout):
               ~~~~~~~~~~~~~~~~^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/timeouts.py", line 116, in __aexit__
    raise TimeoutError from exc_val
TimeoutError
```

Ich bin mir nicht ganz sicher, aber ich hatte mit der anderen extension https://github.com/eddso/ha_sma_speedwire keine connection Probleme. 

Was evtl auch noch interessant ist. Als ich zeitweise beide extension installiert und konfiguriert hatte, funktionierte ha_sma_speedwire plötzlich nicht mehr. Kann es sein das das Problem mehrere gleichzeitige Verbindungsversuche sind?

littleyoda, 2025-02-09:
Ich hatte in der letzten Version bei der Seriennummer die genutzt, die auch [ha_sma_speedwire](https://github.com/eddso/ha_sma_speedwire) nutzt. Das kann zu Problemen führen. In der nächsten Version werde ich wieder eine andere nutzen.

Es gibt aber auch (unbestätigte) Aussagen, dass die Anzahl der Speedwire Verbindungen limitiert sind.

Ansonsten bist du wohl vom dem folgenden Problem betroffen, welches ich aktuell nicht korrigiert bekomme:

https://github.com/littleyoda/ha-pysmaplus/blob/main/docs/faq.md#speedwire-inverter-and-battery-storage

lanwin, 2025-02-09:
Ich hab komischerweise mit der 0.4.7 gerade wieder das es zu einem timeout beim `sensor_def = await sma.get_sensors(entry.data[CONF_DEVICE])`. Testweise habe ich mal den Sunny Explorer gestartet. Der funktioniert ohne Probleme.

lanwin, 2025-02-09:
Sehr merkwürdig. Ich hab jetzt testweise noch mal https://github.com/eddso/ha_sma_speedwire hinzugefügt und konfiguriert und plötzlich gehen beide.

BTW ich bin selbst Software Entwickler. Wenn ich irgendwas testen oder ausprobieren soll, dann sag Bescheid.

littleyoda, 2025-02-09:
Ich befürchte mir ist gerade nicht zu helfen :-)

Ich muss mal von meiner Integration und von ha_sma_speedwire alle Pakete mitloggen.
Sie analysieren, um die Unterschiede zu bestimmen. Ich hoffe, dass ich daraus eine Idee bekommen kann, was schief läuft. 

Wobei:
Ich habe [hier](https://github.com/littleyoda/ha-pysmaplus/issues/39) mal dokumentiert, was ich als erstes auf dem langen Weg benötige.

Danach muss ich dann nochmal den Parser heraus kramen, der mit TCPDUMP zu recht kommt und schauen, ob er auch die ausgehenden Pakete parsen kann.

Wenn das alles funktioniert, kann ich mir die Unterschiede zwischen den beiden Integrationen anschauen.

lanwin, 2025-02-09:
Ich versuche mal ob ich die beiden dumps die Woche erzeugen kann. Kann ich dir die Dumps irgendwie "privat" zukommen lassen. Da ich nicht weiß welche Daten da ausgetauscht werden, möchte ich die nicht für jedermann sichtbar hochladen.

littleyoda, 2025-02-09:
klar:
https://littleyoda.github.io/

danke

staudtm, 2025-02-15:
Ich habe dasselbe Problem: alle SMA Devices sind nach Neustart weg. Ich bin noch ein absoluter HA-Neuling und wollte daher einfach mal fragen, in welcher Datei ich denn die Device-Konfigurationen finde? Dann könnte ich mir die Abschnitte sichern. Oder vielleicht werden ja gar keine Daten abgelegt und das ist die Ursache des Problems(?)

staudtm, 2025-02-15:
Ich habe gerade noch etwas bemerkt: nicht nur nach Neustart, sondern **auch nach einem Backup ist die SMA Integration nicht mehr vorhanden**!?

littleyoda, 2025-02-15:
1. Sie die Geräte nicht mehr vorhanden oder die ganze Integration?
oder anderes gefragt, kannst du so wie es ist, einfach ein neues/altes Gerät wieder hinzufügen?

2. Kannst du bitte mal ein Downgrade zur 0.4.7 probieren?

3. Relevante Dateien sind:
/config/.storage

core.config_entries
Konfigurationsdatei

core.device_entries
Basierend auf der Konfigurationsdatei werden die Geräte anlegt 
(im Fall von pysmaplus sollte es einen Eintrag je Eintrag in der Konfigurationsdatei geben)

core.entity_registry
Hier gibt es für jedes Gerät die verschiedenen Sensoren. 
(für die Dignose weniger Interessant)

staudtm, 2025-02-15:
Ich kann nur kurz antworten, danach bin ich bis Montag "off".
1. In der GUI sehe ich keine Geräte und keine Integration mehr
2. mache ich nächste Woche
3.  

**.storage more core.config_entries|grep sma**
kein Ergebnis

**.storage more core.device_registry|grep sma**
      {"config_entries":["01JM0FQ047FH3AS6ZT1MNK0PD7"],"connections":[],"created_at":"2025-02-13T20:25:09.153249+00:00","identifiers":[["pysmaplus","speedwireinv-3003491886-3003491886"]],"id":"c2204a56edb2d5b8be464680c386f70d","orphaned_timestamp":null,"modified_at":"2025-02-14T21:23:24.963848+00:00"},
      {"config_entries":["01JM4DJYAZ2248PAM94696KF0Z"],"connections":[],"created_at":"2025-02-10T19:57:27.151230+00:00","identifiers":[["pysmaplus","speedwireem-3004907682-3004907682"]],"id":"7fdde7841bf8e2e9acb12829eb7fa052","orphaned_timestamp":null,"modified_at":"2025-02-15T09:18:11.899930+00:00"},
      {"config_entries":["01JM4DPHZR3MT54MZKPF7XG7M5"],"connections":[],"created_at":"2025-02-10T20:00:49.167221+00:00","identifiers":[["pysmaplus","webconnect-3003491886-3003491886"]],"id":"b951575ef370d8dbcc53f4ba42452f5e","orphaned_timestamp":null,"modified_at":"2025-02-15T09:18:11.931383+00:00"},
      {"config_entries":["01JM4DST8T5E8FQA02G83HWT2D"],"connections":[],"created_at":"2025-02-10T20:02:55.114801+00:00","identifiers":[["pysmaplus","webconnect-3017256519-3017256519"]],"id":"e79a93f9b478356e93fc3b9ce9329582","orphaned_timestamp":null,"modified_at":"2025-02-15T09:18:11.971200+00:00"}

**.storage more core.entity_registry |grep sma**
alle entities sind noch da

lanwin, 2025-02-20:
Cool! Was war es denn?

littleyoda, 2025-02-20:
Am Ende war es nur eine falsch initialisierte Variable:
https://github.com/littleyoda/ha-pysmaplus/commit/2e28e09a5d86d433c3d8b9e88095c7c3d7272cf3

hat mich aber am Ende mehr als 2h Fehlersuche über die letzten Tage gekostet.

lanwin, 2025-02-21:
Vielen Dank für deine Zeit!

staudtm, 2025-02-21:
Auch von mir vielen Dank für Deine Arbeit :-)