littleyoda/ha-pysmaplus #37: Energy meter is gone after restarting HA

SanderM2, 2025-02-08:
I think I discovered a bug. I can add my SMA meter using “SMA Energy Meter / Sunny Home Manager 2.0” and it discovers it correctly. It works perfectly fine too.

But when I restart my HA the device is gone. It looks like it’s not saving it or something?

littleyoda, 2025-02-08:
Never heart before of a bug like this.

Can you please go to the integration site in HA.
Is SMA Devices Plus still listed?

[image]

Please check also the logfiles and search for "pysma".

SanderM2, 2025-02-08:
Yes. I have 3 inverters and they stay there after a restart.
Only the energy meter goes missing after a restart.

littleyoda, 2025-02-09:
Do I understand you correctly:
After a reboot, the number of devices show in the integration tab is reduced from 4(?) to 3(?).

Can you please search the log files for ‘pysma’ and see if there are any entries there?

SanderM2, 2025-02-09:
Yes that is correct. Only the SMA Energy Meter goes missing. The others are fine.
Searching in the HA logs I get: No issues found for search term 'pysma'

littleyoda, 2025-02-09:
Can you reproduce the problem?
The administration of the integration is the responsibility of HA. If devices are lost, this is actually a problem for HA.
Do you have the latest version of HA?

SanderM2, 2025-02-09:
I have the latest version of HA and every single time I add the meter it is working perfectly fine until I restart HA.
After a restart I have to add the meter again and it works find again until I restart once again..

I can reproduce this problem over and over

littleyoda, 2025-02-09:
Can you please send me the whole home assistant logfile?

Maybe I'll find something that explains the problem.
https://littleyoda.github.io

littleyoda, 2025-02-10:
@ckgth123 
Did the error occur with a new HomeAssistant version or was the system unchanged?

ckgth123, 2025-02-10:
Hi,

it was an new installation of HA. No update/upgrade.
I set up 1 or more Devices. After an reboot al Devices gone. Till the reboot it works (complete reboot with power down)

littleyoda, 2025-02-10:
The following would be interesting to understand:

- Reboot (note the date & time)
- Add SHM again (note date & time))
- check whether measured values are received
- Reboot (note date & time)
Check whether the device still exists

Please send me the complete log file.
Short-Links to the logfile-tab:
https://my.home-assistant.io/redirect/logs/?

ckgth123, 2025-02-10:
O.K.
at 15:22 HA was reboot.
at 15:27 i have add SHM2,
at 15:28 TripowerX
15:29 checked the measured data. works
15:30 reboot (HA)  - devices are gone

log i have sent.

ckgth123, 2025-02-11:
Hi,

i have tested with older version.v0.4.7. there it works. i think thas a little bug from 0.4.8

regrads
christian

littleyoda, 2025-02-16:
I can confirm hat 4.7.0 is working.
Please downgrade to 4.7.0 for now.

closed 2025-02-20 by littleyoda