littleyoda/ha-pysmaplus #38: nan as value

littleyoda, 2025-02-08:
[A reminder for myself and for the record]
Interface enneox-os

"ValueError: Sensor sensor.sunny_boy_smart_energy_5_0_battery_power_charge_total has device class 'power', state class 'measurement' unit 'W' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-finite value: 'nan'"

Sometimes the inverter returns "nan" (not a number) instant of a measurement.
One possible way of handling this would be to return an unavailable value to HA.

dana-se, 2025-05-18:
@littleyoda 
After latest automatic update to 03.12.15.R on my Tripower X15 inverter, it seems like many sensors, perhaps even all of them, has the non-finite value: 'nan'" when the inverter is idle.

dana-se, 2025-05-18:
Yes, i have the same error message in my home-assistant log.
Is it that information you want?
How do I E-Mail you?

littleyoda, 2025-05-18:
Yes, wanted to make sure it was the same class of error.

E-Mail can be found here:
https://littleyoda.github.io/

littleyoda, 2025-05-18:
Can you test the current beta-version 0.4.8.3?

you have to download the version 0.4.8.3 using hacs.
[image]

[image]

dana-se, 2025-05-18:
Just installed beta-version 0.4.8.3 and will report back tomorrow.

Thanks for fast replay

littleyoda, 2025-05-18:
found another bug.

Please try v0.4.8.4

dana-se, 2025-05-18:
Just installed beta-version 0.4.8.4 and will report back tomorrow.

Thanks for fast replay

dana-se, 2025-05-19:
The errors are still remains

littleyoda, 2025-05-20:
Can you please send me again the diagnostic information.
However, do not generate the diagnostic information until the inverter is actually producing ‘nan’.

dana-se, 2025-05-20:
Here are the files, and the one ending with (1) is after I reloaded the integration.
Hope you find the errors.

dana-se, 2025-05-24:
I think that i have found the problem, according to the diagnostic information, the error was 'NaN' and not 'nan'.
So I'm attaching a file with my changes and hope that you can look at them, and maybe use all or parts of the code for the next version.

The code changes are fixes for
1. 'nan' Errors
2. Warnings: sensor.sunny_tripower_x_15_pv_total_power_a, b and c, with wrong state class

I've now used the code overnight, as error messages used to appear around midnight, but now I don't get any error messages anymore.

Thanks

dana-se, 2025-05-24:
Here is the file

Thanks

"""SMA Solar Webconnect interface."""

from __future__ import annotations

from typing import TYPE_CHECKING

import pysmaplus as pysma

import math

from homeassistant.components.sensor import (
    SensorDeviceClass,
    SensorEntity,
    SensorEntityDescription,
    SensorStateClass,
)
from homeassistant.config_entries import ConfigEntry
from homeassistant.const import (
    PERCENTAGE,
    EntityCategory,
    UnitOfApparentPower,
    UnitOfElectricCurrent,
    UnitOfElectricPotential,
    UnitOfEnergy,
    UnitOfFrequency,
    UnitOfPower,
    UnitOfTemperature,
    UnitOfReactivePower,
)
from homeassistant.core import HomeAssistant
from homeassistant.helpers.device_registry import DeviceInfo
from homeassistant.helpers.entity_platform import AddEntitiesCallback
from homeassistant.helpers.typing import StateType
from homeassistant.helpers.update_coordinator import (
    CoordinatorEntity,
    DataUpdateCoordinator,
)

from .const import (
    DOMAIN,
    PYSMA_COORDINATOR,
    PYSMA_DEVICE_INFO,
    PYSMA_SENSORS,
    PYSMA_ENTITIES,
)

SENSOR_ENTITIES: dict[str, SensorEntityDescription] = {
    "status": SensorEntityDescription(
        key="status",
        name="Status",
        entity_category=EntityCategory.DIAGNOSTIC,
    ),
    "operating_status_general": SensorEntityDescription(
        key="operating_status_general",
        name="Operating Status General",
        entity_registry_enabled_default=False,
        entity_category=EntityCategory.DIAGNOSTIC,
    ),
    "inverter_condition": SensorEntityDescription(
        key="inverter_condition",
        name="Inverter Condition",
        entity_registry_enabled_default=False,
        entity_category=EntityCategory.DIAGNOSTIC,
    ),
    "inverter_system_init": SensorEntityDescription(
        key="inverter_system_init",
[... 956 lines cut ...]
        ):
            name_prefix = "SMA"

        return f"{name_prefix} {super().name}"

    @property
    def native_value(self) -> StateType:
        """Return the state of the sensor."""
        value = self._sensor.value
        if isinstance(value, str) and value.lower() == "nan":
            return None
        if isinstance(value, float):
            if math.isnan(value) or not math.isfinite(value):
                return None
        return value

    async def async_added_to_hass(self) -> None:
        """Run when entity about to be added to hass."""
        await super().async_added_to_hass()
        self._sensor.enabled = True

    async def async_will_remove_from_hass(self) -> None:
        """Run when entity will be removed from hass."""
        await super().async_will_remove_from_hass()
        self._sensor.enabled = False

littleyoda, 2025-05-24:
Thanks. Uploading Files via E-Mail does not work  for github

dana-se, 2025-05-24:
What should I do then?

littleyoda, 2025-05-24:
You can upload the files on the github-Website (https://github.com/littleyoda/ha-pysmaplus/issues/38)  or send it by e-mail to me directly.

dana-se, 2025-05-24:
> For now, I have released the v0.4.8.5 as a beta-version with your fixes for nan.
>
> I have excluded the state_class changes. I would like to take a closer look at this topic. As I had already changed the class once because of a warning.
>
> Thanks a lot.

Here are my thoughts on state_class changes.

According the state_class change, see the warnings in the attached log.

In  version 0.4.8.2 you changed device_class from POWER to ENERGY.
And if device_class is ENERGY, then state_class must bee TOTAL, TOTAL_INCREASING or NONE for no LTS.

Thanks

closed 2025-06-10 by littleyoda