littleyoda/ha-pysmaplus #42: Disconnecting Sunny Home Manager 2 connection

Stefan-83, 2025-02-14:
Used Version of Home Assistant: 2025.2.3
Used Version of ha-pysmaplus: 0.4.8.1

**Describe the bug, Remarks, ...**
After the discovery of the Sunny Home Manager 2, the integration works like expected and is showing the correct data:

[image]

**Used Interface/Access Method**
Energymeter/SHM2: [x]

**List of the integrated SMA devices**
Inverter: 
Sunny Home Manager 2: Firmware: 2.15.7.R

**Screenshots and Diagnose Information**
If applicable, add screenshots to help explain your problem and send the [Diagnose Information](https://raw.githubusercontent.com/littleyoda/ha-pysmaplus/main/docs/Diagnose-Information.png) to the developer. (E-Mail can be found here: https://littleyoda.github.io/)

littleyoda, 2025-02-14:
I'm not understanding your Bugreport.

Is this related to:
https://github.com/littleyoda/ha-pysmaplus/issues/37

Stefan-83, 2025-02-14:
> I'm not understanding your Bugreport.
> 
> Is this related to: [#37](https://github.com/littleyoda/ha-pysmaplus/issues/37)

sorry, I accidentally clicked on post and edited now :)

littleyoda, 2025-03-02:
It looks like a massive network problem.

I have put together a list of common problems here:
https://github.com/littleyoda/ha-pysmaplus/blob/main/docs/shm2_multicast.md