littleyoda/ha-pysmaplus #44: v.4.8.2 untersützt nicht alle Entitäten des Wechselrichters SMA STP3.0-3AV-40

JuergenK53, 2025-03-06:
Used Version of Home Assistant: aktuelle Version
Used Version of ha-pysmaplus: v.4.8.2

**Describe the bug, Remarks, ...**
A clear and concise description of what the bug is.
Mit der Version v.4.8.1 wurden alle Entitäten meines SMA-Wechselrichters STP3.0-3AV-40 richtig angezeigt.
Mit der neuen Version v.4.8.2 fehlt ein Großteil der Entitäten.

**Used Interface/Access Method**
see https://raw.githubusercontent.com/littleyoda/ha-pysmaplus/main/docs/images/interfaces.png
Speedwire: [x]
Webconnect: [ ]
EnnexOS-based: [ ]
Energymeter/SHM2: [ ]
SHM2 with Grid Guard Code: [ ]

**List of the integrated SMA devices**
Inverter: STP3.0-3AV-40
Energymeter:
Sunny Home Manager 2:

**Screenshots and Diagnose Information**
If applicable, add screenshots to help explain your problem and send the [Diagnose Information](https://raw.githubusercontent.com/littleyoda/ha-pysmaplus/main/docs/Diagnose-Information.png) to the developer. (E-Mail can be found here: https://littleyoda.github.io/)

premier-lifeboat, 2025-05-26:
the same for my STP8.0-3AV-40
in the end there is no data provided at all: either they are "deactivated" or "not provided" (nicht bereitgestellt)

after updating to v4.8.2 I initially had seen an error after reloading the device - which disappeared after the second try

2025-05-26 15:46:07.680 INFO (MainThread) [custom_components.pysmaplus] Creating new pysma-Device [Session: False, Verify-SSL: False, Url: 192.168.xxx.xxx, User/Group: user, Accessmethod: speedwireinv]
2025-05-26 15:46:09.693 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry STP8.0-3AV-40 (Sunny Tripower 8.0) (sw3006550xxx) for pysmaplus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 749, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/pysmaplus/__init__.py", line 103, in async_setup_entry
    sma = await getPysmaInstance(hass, entry.data)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/pysmaplus/__init__.py", line 96, in getPysmaInstance
    await sma.new_session()
  File "/usr/local/lib/python3.13/site-packages/pysmaplus/device_speedwire.py", line 483, in new_session
    await self.device_info()
  File "/usr/local/lib/python3.13/site-packages/pysmaplus/device_speedwire.py", line 492, in device_info
    ll = await self.device_list()
         ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pysmaplus/device_speedwire.py", line 503, in device_list
    await asyncio.wait_for(fut, timeout=self._protocol._overallTimeout)
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 507, in wait_for
    return await fut
           ^^^^^^^^^
pysmaplus.exceptions.SmaConnectionException: Login failed! No Response from Device!

iroc-fabi, 2025-06-01:
Ich nutze auch die Version 4.8.2, mir fehlt die Temperatur des Wechselrichters STP5000. War die denn vorher da? Wie kann ich zum testen die neuste Version 4.8.5 installieren?