littleyoda/ha-pysmaplus #46: MQTT error - The unit of measurement `VAr` is not valid together with device class `energy`

Johnny-Nova, 2025-03-08:
Used Version of Home Assistant: 2025.3.1
Used Version of ha-pysmaplus: 0.4.8.2

**Describe the bug, Remarks, ...**
Following error in log file:
Logger: homeassistant.components.mqtt.entity
Source: components/mqtt/entity.py:155
integration: MQTT (documentation, issues)
First occurred: 11:19:15 (5 occurrences)
Last logged: 11:19:15

Error 'The unit of measurement `VAr` is not valid together with device class `energy`' when processing MQTT discovery message topic: 'homeassistant/sensor/3017578383/qconsume/config', message: '{'state_class': 'total_increasing', 'state_topic': 'SMA-EM/status/3017578383/qconsume', 'unique_id': '3017578383_qconsume', 'name': 'qconsume', 'unit_of_measurement': 'VAr', 'device': {'model': 'Energy Meter', 'identifiers': ['sma_em_3017578383'], 'name': 'SMA Energy Meter', 'manufacturer': 'SMA'}, 'device_class': 'energy'}'
Error 'The unit of measurement `VAr` is not valid together with device class `energy`' when processing MQTT discovery message topic: 'homeassistant/sensor/3017578383/qsupply/config', message: '{'state_class': 'total_increasing', 'state_topic': 'SMA-EM/status/3017578383/qsupply', 'unique_id': '3017578383_qsupply', 'name': 'qsupply', 'unit_of_measurement': 'VAr', 'device': {'model': 'Energy Meter', 'identifiers': ['sma_em_3017578383'], 'name': 'SMA Energy Meter', 'manufacturer': 'SMA'}, 'device_class': 'energy'}'
Error 'The unit of measurement `VA` is not valid together with device class `energy`' when processing MQTT discovery message topic: 'homeassistant/sensor/3017578383/sconsume/config', message: '{'state_class': 'total_increasing', 'state_topic': 'SMA-EM/status/3017578383/sconsume', 'unique_id': '3017578383_sconsume', 'name': 'sconsume', 'unit_of_measurement': 'VA', 'device': {'model': 'Energy Meter', 'identifiers': ['sma_em_3017578383'], 'name': 'SMA Energy Meter', 'manufacturer': 'SMA'}, 'device_class': 'energy'}'
Error 'The unit of measurement `VA` is not valid together with device class `energy`' when processing MQTT discovery message topic: 'homeassistant/sensor/3017578383/ssupply/config', message: '{'state_class': 'total_increasing', 'state_topic': 'SMA-EM/status/3017578383/ssupply', 'unique_id': '3017578383_ssupply', 'name': 'ssupply', 'unit_of_measurement': 'VA', 'device': {'model': 'Energy Meter', 'identifiers': ['sma_em_3017578383'], 'name': 'SMA Energy Meter', 'manufacturer': 'SMA'}, 'device_class': 'energy'}'
Error 'The unit of measurement `A` is not valid together with device class `energy`' when processing MQTT discovery message topic: 'homeassistant/sensor/3017578383/i1/config', message: '{'state_class': 'total_increasing', 'state_topic': 'SMA-EM/status/3017578383/i1', 'unique_id': '3017578383_i1', 'name': 'i1', 'unit_of_measurement': 'A', 'device': {'model': 'Energy Meter', 'identifiers': ['sma_em_3017578383'], 'name': 'SMA Energy Meter', 'manufacturer': 'SMA'}, 'device_class': 'energy'}'

**Used Interface/Access Method**
see https://raw.githubusercontent.com/littleyoda/ha-pysmaplus/main/docs/images/interfaces.png
Speedwire: [ ]
Webconnect: [ ]
EnnexOS-based: [ ]
Energymeter/SHM2: [X]
SHM2 with Grid Guard Code: [ ]

List of the integrated SMA devices
Inverter: Sunny Island 4.4M-11 (sw1261043877), EVC7.4-1AC-10 (en3014279955)
Energymeter:
Sunny Home Manager 2: HM 2.0 (em3017578383)

**Screenshots and Diagnose Information**
If applicable, add screenshots to help explain your problem and send the [Diagnose Information](https://raw.githubusercontent.com/littleyoda/ha-pysmaplus/main/docs/Diagnose-Information.png) to the developer. (E-Mail can be found here: https://littleyoda.github.io/)

KCarlsen, 2025-04-28:
there is a feature request to add reactive power i HA. 
https://community.home-assistant.io/t/can-we-please-add-device-class-reactive-energy-in-sensor/774319

littleyoda, 2025-04-30:
Johnny-Nova, that Integration do you use?

Because your bug is related to mqtt it can not be ha-pysmaplus.

dropqube, 2025-05-18:
@littleyoda 

Is this a similiar problem? I have *not* changed the entity manually - *NOT* MQTT related
This is not a single occurance, its logged pretty often

Logger: homeassistant.components.sensor
Source: components/sensor/__init__.py:548
integration: Sensor (documentation, issues)
First occurred: May 17, 2025 at 18:48:44 (3 occurrences)
Last logged: May 17, 2025 at 18:48:44

Entity sensor.sunny_tripower_x_15_pv_total_power_a (<class 'custom_components.pysmaplus.sensor.SMAsensor'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total', 'total_increasing'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/littleyoda/ha-pysmaplus/issues
Entity sensor.sunny_tripower_x_15_pv_total_power_b (<class 'custom_components.pysmaplus.sensor.SMAsensor'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total', 'total_increasing'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/littleyoda/ha-pysmaplus/issues
Entity sensor.sunny_tripower_x_15_pv_total_power_c (<class 'custom_components.pysmaplus.sensor.SMAsensor'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total', 'total_increasing'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/littleyoda/ha-pysmaplus/issues

closed 2025-05-18 by littleyoda