littleyoda/ha-pysmaplus #50: Error setting up entry Sunny Home Manager 2 after Update to HAAS 2025.4.2

Kragothius, 2025-04-13:
Used Version of Home Assistant: 2025.4.2
Used Version of ha-pysmaplus: 4.8.2

**Describe the bug, Remarks, ...**
This is more a Warning. 
I just updated from HASS 2025.4.1 zu 2025.4.2 and got issues with the Connection to SHM2 and 

Error in the Log says: 
Error setting up entry Sunny Home Manager 2 (emxxxxxxxx) for pysmaplus
2025-04-13 11:39:47.689 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry Sunny Home Manager 2 (emxxxxxxx) for pysmaplus
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 507, in wait_for
    return await fut
           ^^^^^^^^^
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 751, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/pysmaplus/__init__.py", line 108, in async_setup_entry
    device_list = await sma.device_list()
                  ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pysmaplus/device_em.py", line 119, in device_list
    data = await self._get_next_values()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pysmaplus/device_em.py", line 92, in _get_next_values
    await asyncio.wait_for(self._data_received, timeout=timeout)
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 506, in wait_for
    async with timeouts.timeout(timeout):
               ~~~~~~~~~~~~~~~~^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/timeouts.py", line 116, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

After reverting to 2025.4.1 everything is working fine again.

**Used Interface/Access Method**
see https://raw.githubusercontent.com/littleyoda/ha-pysmaplus/main/docs/images/interfaces.png
Speedwire: [ ]
Webconnect: [ X]
EnnexOS-based: [ X]
Energymeter/SHM2: [ X]
SHM2 with Grid Guard Code: [ ]

**List of the integrated SMA devices**
Inverter: Sunny Tripower 4 and Sunny Tripower 5
Energymeter: SHM2
Sunny Home Manager 2: YES

**Screenshots and Diagnose Information**
If applicable, add screenshots to help explain your problem and send the [Diagnose Information](https://raw.githubusercontent.com/littleyoda/ha-pysmaplus/main/docs/Diagnose-Information.png) to the developer. (E-Mail can be found here: https://littleyoda.github.io/)

littleyoda, 2025-04-13:
I updated to 4.8.2 and SHM2 runs without a problem.

Can you please search in the logfiles for "pysma".

The message above only means that the integration is no longer receiving messages from SHM2.

Kragothius, 2025-04-13:
As i reverted to 2025.4.1 is do not have any logs by now.
But I will do another rollout for 2025.4.2 and see what happens.
If the Problem occurs again, I will search the logs.

Kragothius, 2025-04-13:
Just did the Update again. Everything is working as expected. 
Closed

closed 2025-04-13 by Kragothius