littleyoda/ha-pysmaplus #53: Able to control Grid charging on STP (WebConnect)

KCarlsen, 2025-04-22:
Hi

To be able to control when to charge battery from grid on the STP hybrid inverter you can control the 'Minimum width of backup power area' value by setting it to e.g 100% when you want to charge battery from grid in cheap periods.
I assume its the same the SMA home manager 2 does, but i don't have that, so i tried myself.

Please make this available in this awesome plugin, so we can get/set it through webconnect

The setParamValues payload ( set it to 40%) is `{"destDev":[],"values":[{"6802_00895100":{"9":[40]}}]}`

Thanks

KCarlsen, 2025-05-06:
its called ''Minimum width of backup power area' in the inverter. 
Its described here: https://manuals.sma.de/SI-13/en-US/1414774155.html (range C)

KimCarlsen, 2025-10-05:
Not to rush for this, but is there anything i can assist on, regarding this? or buy some coffee
Sun is not that frequent a guest so with be nice to be able to have battery charged from grid to use when high prices.

zlFamous, 2026-04-08:
@KimCarlsen can you please let me know how you done that?