littleyoda/ha-pysmaplus #54: Restarting HA/PYSMA during night causes sensors to be permanent unavailable

KNXBroker, 2025-04-27:
Used Version of Home Assistant: 2025.4.4
Used Version of ha-pysmaplus: 0.4.8.2

**Describe the bug, Remarks, ...**
I have connected STP 6000TL-20 via SpeedWire. If I reload the integration or restart home assistant during day (inverter active) everything is fine. All sensors are created. However if I reload during night or restart Home Assistant during night, some sensor are stated unavailable.  Unfortunately they stay unavailalbe, also during the next day. Only a restart / reload fixes the  issue.

A clear and concise description of what the bug is.

**Used Interface/Access Method**
see https://raw.githubusercontent.com/littleyoda/ha-pysmaplus/main/docs/images/interfaces.png
Speedwire: [x]
Webconnect: [ ]
EnnexOS-based: [ ]
Energymeter/SHM2: [ ]
SHM2 with Grid Guard Code: [ ]

**List of the integrated SMA devices**
Inverter: STP 6000TL-20

**Screenshots and Diagnose Information**
Note: Sensor Störung is a custom template
[image]

If applicable, add screenshots to help explain your problem and send the [Diagnose Information](https://raw.githubusercontent.com/littleyoda/ha-pysmaplus/main/docs/Diagnose-Information.png) to the developer. (E-Mail can be found here: https://littleyoda.github.io/)

KNXBroker, 2025-05-06:
Thank you for your reply, I know other inverter integrations behaviour similar, e.g. fronius, but fronius make the sensors available once present (do not know if this information is helpful).

I think I will step back to my direct modbus solution, as I am often updating in the evening. 

Short term for other people could be a manual reload of the integration once daylight is starting

Nevertheless impressive work you are doing here! Thumbs pressed for a solution one day.

KNXBroker, 2025-08-29:
> The STP6.0SE (SUNNY TRIPOWER 6.0 SE) hybrid seems not supported with SMA Devices Plus. HA can not read data, nor write to ModBus registers to force loading BYD batteries, or unload them.

Not releated to this issue, right?

littleyoda, 2025-10-20:
Unfortunately, I have no idea how to get around this problem.

Planetexpress3k, 2025-10-20:
I'd love to help, but my programming skills are absolutely limited. 
is there any change to check internally for "not available" sensors during initialization und start a re-initialization process every minute or so?

alecowiter, 2026-04-07:
> > Unfortunately, I have no idea how to get around this problem.
> 
> I'm not sure if you're aware of this program: [SBFspot](https://github.com/SBFspot)?
> 
> It is used to pull values from speedwire inverters and the command line program has got an option to force an inquiry (-finq) even after sundown (works day and night). Just works just fine and a similar implementation could be the solution.
> 
> Unfortunately I'm no programmer, so I can not tell you where this options points to in de code... But maybe someone else can of maybe one of the maintainers of SBFspot is willing to help?

Thanks for a nice suggestion. I checked the code and found this commit:
https://github.com/SBFspot/SBFspot/commit/feaaa5d3a96957bfe0c4936a44fd955808de2f23
I do hope that this is implementable in the current repository. 
Looking forward for the reply of the maintainers.