littleyoda/ha-pysmaplus #55: {"message":"invalid handler specified"}

littleyoda, 2025-05-05:
> I seem to have exactly that problem: {"message":"invalid handler specified"}
> Inverter Tripower X 15
> Energy meter
> Sunny Home Manager 2 with current Firmware
> 
> What can I do to help - I am not experienced at all with HA, but I am able to give hints on SMA devices ;-)  

 _Originally posted by @raimund-hdr in [#40](https://github.com/littleyoda/ha-pysmaplus/issues/40#issuecomment-2700528085)_

littleyoda, 2025-05-05:
@raimund-hdr
@DonKedero
Can you both check the logfiles (please search for pysma).
Unfortunately, I cannot reproduce the error.

DonKedero, 2025-05-06:
@littleyoda I can't reproduce the issue as described here (https://github.com/littleyoda/ha-pysmaplus/issues/40#issuecomment-2850130447)

I've removed SMA Devices Plus from HACS, restarted my HA, downloaded v0.4.8.2 again, restarted my HA, tried to configure the integration again ... no error (besides the issue I already had)

guessing this is a one-time off ... and not reproducable :/

serge402, 2025-05-06:
Nope not a one off.
I get "Config flow could not be loaded: {"message":"Invalid handler specified"}" when i try to start the addon.

Logs:
Logger: homeassistant.config_entries
Source: config_entries.py:3771
First occurred: 9:00:13 PM (1 occurrences)
Last logged: 9:00:13 PM

Error occurred loading flow for integration pysmaplus: No module named 'custom_components.pysmaplus.helper'

DonKedero, 2025-05-06:
@serge402 is this your first install of SMA Devices Plus? Is your version v0.4.8.2?
Can you get soms logs? Because I've done a downgrade towards v0.4.7 and upgrade towards v0.4.8.2 which resolves the issue (with me) ... but I can't reproduce it anymore even after a removal.

serge402, 2025-05-06:
@littleyoda What I’m really hoping to access is any data that indicates when the inverters are being throttled (curtailed) and ideally by how much.

My setup includes a Home Manager 2.0, with two inverters: a Sunny Boy SB5000TL-21 and an SB5.0-1AV-41.

I completely understand the limitations due to SMA’s closed protocol—huge respect to those who’ve managed to reverse-engineer as much as they have already. If you have any insights, even partial, on whether this kind of curtailment information might be exposed somewhere (or how it could be inferred), I’d really appreciate it!

littleyoda, 2025-05-06:
No. I am not aware of any solution. The SHM2 should actually provide the information, but the information available from the SHM2 is very limited. But I have never read about an approach for the different device generations either.

Is the information available anywhere at SMA (e.g. in the portal)?

serge402, 2025-05-06:
@littleyoda No sign of curtailment status or limits in Sunny Portal or local web interfaces. The inverter production just drops, which makes it really difficult to distinguish between curtailment and normal fluctuations due to clouds or consumption shifts.

I mainly run two SB5000TL-21s myself, but I’ve had access to a newer model for testing. I was hoping that maybe a newer inverter would expose more detailed metrics—especially when combined with the SHM2—but so far I haven’t seen anything conclusive.

littleyoda, 2025-05-06:
Moved the topic to the discussion area:
https://github.com/littleyoda/ha-pysmaplus/discussions/56