littleyoda/ha-pysmaplus #58: Change password of already established SMA entity

stoffelll, 2025-05-22:
Used Version of Home Assistant: 2025.05.
Used Version of ha-pysmaplus: 0.482

**Update/Change password of an existing SMA-entity/device**
I had to change the user password of my SMA inverter and didn't found a way to update it in the integration.
So finally I had to delete it and re-add the SMA entity/device with the new password.

littleyoda, 2025-05-25:
There is currently no function to change the password => Feature Request

Sl0ter, 2025-07-29:
If you go to the core.config_entries in config/.storage you can see the clear password. (in speedwire, i don't know if it's the same with the other methods )
there you can change it.

littleyoda, 2025-07-30:
Password are stored always unencrypted in HomeAssistant.
See other integrations..