littleyoda/ha-pysmaplus #61: Problem following update 0.4.9

Nilan464, 2025-06-10:
Hi,

Home Assistant is unable to reload my SUNNY TRIPOWER X 15 integration since the last update to 0.4.9.
Has anyone else had this problem too?
I'm not sure what to do, but I'd like to avoid having to restart from an old backup that would cause me to lose a lot of things. :(

Here is what the logs tell me

```
Enregistreur: homeassistant.config_entries
Source: config_entries.py:749
S'est produit pour la première fois: 18:37:05 (2 occurrences)
Dernier enregistrement: 18:37:55

Error setting up entry SUNNY TRIPOWER X 15 (sw3022914200) for pysmaplus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 749, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/pysmaplus/__init__.py", line 109, in async_setup_entry
    sma_device_info = device_list[entry.data[CONF_DEVICE]]
                      ~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^
KeyError: '3022914200'
```

Thanks

littleyoda, 2025-06-10:
1. You can always downgrade to an older version of the integration using HACS. No need to restore from a backup.
 
2. Have you tried to reload Home Assistant or have you simply tried to reload the integration?

markus5790, 2025-06-10:
After the update I cannot connect to my SMA Sunny Tripower 5000 TL-20 anymore. Even if I downgrade to an older/previous version, the connection cannot be established anymore. Please help.

littleyoda, 2025-06-10:
> After the update I cannot connect to my SMA Sunny Tripower 5000 TL-20 anymore. Even if I downgrade to an older/previous version, the connection cannot be established anymore. Please help.

This bug is related to this:
https://github.com/littleyoda/ha-pysmaplus/blob/main/docs/speedwire.md

Unfortunately, I don't have a solution for this. The problem is not reproducible. It can appear suddenly and then suddenly disappear again.

markus5790, 2025-06-10:
I was reading through some issues in your project and found this integration: https://github.com/Wired-Square/homeassistant-sma-sw
with this integration connection to the inverter works fine again.

markus5790, 2025-06-10:
The Inverter is also accessible via the evcc Home Assistant add-on and its speedwire plugin

Nilan464, 2025-06-10:
I found this in the logs.
It appears at the same time as the log mentioned in the first post.
Perhaps this can help understand what's going on?

```
Enregistreur: pysmaplus.device_speedwire
Source: runner.py:154
S'est produit pour la première fois: 22:15:35 (30 occurrences)
Dernier enregistrement: 23:04:12

Skipping message. --- Len b"SMA\x00\x00\x04\x02\xa0\x00\x00\x00\x01\x00\x9e\x00\x10`e'\xf0}\x00#\x19\x02#\x00\x00\xab\x01\x98\x02.\xb4\x00\x00\x00\x00\x00\x00\x02\x80\x01\x02\x00X\x00\x00\x00\x00\x02\x00\x00\x00\x01\x1e\x82\x10i\x97HhOnduleur SMA STP 15-50\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x1f\x82\x08i\x97HhA\x1f\x00\x01\xfe\xff\xff\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01 \x82\x08j\x97Hh\x13%\x00\x01\xfe\xff\xff\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00" Ril 8527360 33554433 x -1 bytes
Skipping message. --- Len b'SMA\x00\x00\x04\x02\xa0\x00\x00\x00\x01\x00.\x00\x10`e\x0b\xf0}\x00#\x19\x02#\x00\x01\xab\x01\x98\x02.\xb4\x00\x01\x00\x00\x00\x00\x05\x80\r\x04\xfd\xff\n\x00\x00\x00\x84\x03\x00\x00\xcb\x9dHh\x00\x00\x00\x00\x00\x00\x00\x00' Ril 4758784 2047016961 x -1 bytes
Skipping message. --- Len b"SMA\x00\x00\x04\x02\xa0\x00\x00\x00\x01\x00\x9e\x00\x10`e'\xf0}\x00#\x19\x02#\x00\x00\xab\x01\x98\x02.\xb4\x00\x00\x00\x00\x00\x00\x06\x80\x01\x02\x00X\x00\x00\x00\x00\x02\x00\x00\x00\x01\x1e\x82\x10i\x97HhOnduleur SMA STP 15-50\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x1f\x82\x08i\x97HhA\x1f\x00\x01\xfe\xff\xff\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01 \x82\x08j\x97Hh\x13%\x00\x01\xfe\xff\xff\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00" Ril 8527360 33554433 x -1 bytes
Skipping message. --- Len b'SMA\x00\x00\x04\x02\xa0\x00\x00\x00\x01\x00.\x00\x10`e\x0b\xf0}\x00#\x19\x02#\x00\x01\xab\x01\x98\x02.\xb4\x00\x01\x00\x00\x00\x00\t\x80\r\x04\xfd\xff\n\x00\x00\x00\x84\x03\x00\x00\xcc\x9dHh\x00\x00\x00\x00\x00\x00\x00\x00' Ril 4758784 2047016961 x -1 bytes
Skipping message. --- Len b"SMA\x00\x00\x04\x02\xa0\x00\x00\x00\x01\x00\x9e\x00\x10`e'\xf0}\x00#\x19\x02#\x00\x00\xab\x01\x98\x02.\xb4\x00\x00\x00\x00\x00\x00\n\x80\x01\x02\x00X\x00\x00\x00\x00\x02\x00\x00\x00\x01\x1e\x82\x10i\x97HhOnduleur SMA STP 15-50\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x1f\x82\x08i\x97HhA\x1f\x00\x01\xfe\xff\xff\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01 \x82\x08j\x97Hh\x13%\x00\x01\xfe\xff\xff\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00" Ril 8527360 33554433 x -1 bytes
```

marky5525, 2025-06-11:
Same here:
`2025-06-11 07:35:06.603 INFO (MainThread) [custom_components.pysmaplus] Creating new pysma-Device [Session: False, Verify-SSL: False, Url: 192.168.178.48, User/Group: user, Accessmethod: speedwireinv]
2025-06-11 07:35:06.603 ERROR (MainThread) [custom_components.pysmaplus] Login retries: 3
2025-06-11 07:35:06.664 WARNING (MainThread) [pysmaplus.device_speedwire] Skipping message. --- Len b"SMA\x00\x00\x04\x02\xa0\x00\x00\x00\x01\x00:\x00\x10`e\x0e\xd0}\x00#\x19\x02#\x00\x01\xb5\x00#\xe6'\x12\x00\x01\x00\x00\x00\x00\x01\x80\r\x04\xfd\xff\x07\x00\x00\x00\x84\x03\x00\x00\x8a\x15Ih\x00\x00\x00\x00\xde\xe9\xfa\xe9\xbc\xbd\xbb\xbd\x88\x88\x88\x88\x00\x00\x00\x00" Ril 4789504 2097348609 x -1 bytes
2025-06-11 07:35:06.692 WARNING (MainThread) [pysmaplus.device_speedwire] Skipping message. --- Len b"SMA\x00\x00\x04\x02\xa0\x00\x00\x00\x01\x00\xc6\x00\x10`e1\x90}\x00#\x19\x02#\x00\xa0\xb5\x00#\xe6'\x12\x00\x00\x00\x00\x00\x00\x02\x80\x01\x02\x00X\x00\x00\x00\x00\x03\x00\x00\x00\x01\x1e\x82\x10\x90\xf6HhSN: 304604707\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x1f\x82\x08\x90\xf6HhA\x1f\x00\x01\xfe\xff\xff\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01 \x82\x08\x90\xf6Hh\x8a#\x00\x00\x8b#\x00\x00\x8c#\x00\x01\x8e#\x00\x00\x8f#\x00\x00A$\x00\x00B$\x00\x00C$\x00\x00\x01 \x82\x08\x90\xf6Hh\xfe\xff\xff\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00" Ril 8527360 50331649 x -1 bytes
2025-06-11 07:35:07.134 INFO (MainThread) [custom_components.pysmaplus] Update pysma localhost/speedwireem/3009869581
2025-06-11 07:35:07.172 WARNING (MainThread) [pysmaplus.device_speedwire] Skipping message. --- Len b"SMA\x00\x00\x04\x02\xa0\x00\x00\x00\x01\x00:\x00\x10`e\x0e\xd0}\x00#\x19\x02#\x00\x01\xb5\x00#\xe6'\x12\x00\x01\x00\x00\x00\x00\x05\x80\r\x04\xfd\xff\x07\x00\x00\x00\x84\x03\x00\x00\x8b\x15Ih\x00\x00\x00\x00\xde\xe9\xfa\xe9\xbc\xbd\xbb\xbd\x88\x88\x88\x88\x00\x00\x00\x00" Ril 4789504 2097348609 x -1 bytes
2025-06-11 07:35:07.194 WARNING (MainThread) [pysmaplus.device_speedwire] Skipping message. --- Len b"SMA\x00\x00\x04\x02\xa0\x00\x00\x00\x01\x00\xc6\x00\x10`e1\x90}\x00#\x19\x02#\x00\xa0\xb5\x00#\xe6'\x12\x00\x00\x00\x00\x00\x00\x06\x80\x01\x02\x00X\x00\x00\x00\x00\x03\x00\x00\x00\x01\x1e\x82\x10\x90\xf6HhSN: 304604707\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x1f\x82\x08\x90\xf6HhA\x1f\x00\x01\xfe\xff\xff\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01 \x82\x08\x90\xf6Hh\x8a#\x00\x00\x8b#\x00\x00\x8c#\x00\x01\x8e#\x00\x00\x8f#\x00\x00A$\x00\x00B$\x00\x00C$\x00\x00\x01 \x82\x08\x90\xf6Hh\xfe\xff\xff\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00" Ril 8527360 50331649 x -1 bytes
2025-06-11 07:35:07.608 DEBUG (MainThread) [custom_components.pysmaplus] Finished fetching pysmaplus data in 0.473 seconds (success: True)
2025-06-11 07:35:07.652 WARNING (MainThread) [pysmaplus.device_speedwire] Skipping message. --- Len b"SMA\x00\x00\x04\x02\xa0\x00\x00\x00\x01\x00:\x00\x10`e\x0e\xd0}\x00#\x19\x02#\x00\x01\xb5\x00#\xe6'\x12\x00\x01\x00\x00\x00\x00\t\x80\r\x04\xfd\xff\x07\x00\x00\x00\x84\x03\x00\x00\x8b\x15Ih\x00\x00\x00\x00\xde\xe9\xfa\xe9\xbc\xbd\xbb\xbd\x88\x88\x88\x88\x00\x00\x00\x00" Ril 4789504 2097348609 x -1 bytes
2025-06-11 07:35:07.682 WARNING (MainThread) [pysmaplus.device_speedwire] Skipping message. --- Len b"SMA\x00\x00\x04\x02\xa0\x00\x00\x00\x01\x00\xc6\x00\x10`e1\x90}\x00#\x19\x02#\x00\xa0\xb5\x00#\xe6'\x12\x00\x00\x00\x00\x00\x00\n\x80\x01\x02\x00X\x00\x00\x00\x00\x03\x00\x00\x00\x01\x1e\x82\x10\x90\xf6HhSN: 304604707\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x1f\x82\x08\x90\xf6HhA\x1f\x00\x01\xfe\xff\xff\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01 \x82\x08\x90\xf6Hh\x8a#\x00\x00\x8b#\x00\x00\x8c#\x00\x01\x8e#\x00\x00\x8f#\x00\x00A$\x00\x00B$\x00\x00C$\x00\x00\x01 \x82\x08\x90\xf6Hh\xfe\xff\xff\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00" Ril 8527360 50331649 x -1 bytes
2025-06-11 07:35:08.120 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry STP 7000TL-20 (sw304604707) for pysmaplus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 749, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/pysmaplus/__init__.py", line 109, in async_setup_entry
    sma_device_info = device_list[entry.data[CONF_DEVICE]]
                      ~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^
KeyError: '304604707'
2025-06-11 07:35:12.213 INFO (MainThread) [custom_components.pysmaplus] Update pysma localhost/speedwireem/3009869581
2025-06-11 07:35:12.604 DEBUG (MainThread) [custom_components.pysmaplus] Finished fetching pysmaplus data in 0.391 seconds (success: True)
2025-06-11 07:35:17.125 INFO (MainThread) [custom_components.pysmaplus] Update pysma localhost/speedwireem/3009869581
2025-06-11 07:35:17.607 DEBUG (MainThread) [custom_components.pysmaplus] Finished fetching pysmaplus data in 0.482 seconds (success: True)
2025-06-11 07:35:22.119 INFO (MainThread) [custom_components.pysmaplus] Update pysma localhost/speedwireem/3009869581
2025-06-11 07:35:22.603 DEBUG (MainThread) [custom_components.pysmaplus] Finished fetching pysmaplus data in 0.484 seconds (success: True)
`

Mark481971, 2025-06-11:
Hello together,

I have the same problem with my inverters STP 150000 and also the battery inverter SBS5. I tried downgrading to the previous version from HACS, but the problem persists.

marky5525, 2025-06-11:
As I See, this issue is not related to the update. the Integration stopped providing the values with the previous version of the software.

Nilan464, 2025-06-11:
Ok, I can set up a VM with SSH access for you.
Do you also need a Home Assistant instance for testing?
Can we communicate privately to discuss the details?

BramVandendaele, 2025-06-11:
Hi I am not sure if this is related to this issue. But since a few days my connection doesn't work as well. 

The integration sensors became unavailable and when I tried to delete and reconfigure I receive this message (see screenshot). 
[image]

When I click on add it says "Unknown error occured"

Some logs: 

`Logger: pysmaplus.device_speedwire
Source: runner.py:154
First occurred: 09:47:41 (8 occurrences)
Last logged: 09:59:46

Skipping message. --- Len b'SMA\x00\x00\x04\x02\xa0\x00\x00\x00\x01\x00\xc6\x00\x10`e1\x90}\x00#\x19\x02#\x00\xa0\x8a\x00*\xe9\xf8~\x00\x00\x00\x00\x00\x00\x02\x80\x01\x02\x00X\x00\x00\x00\x00\x03\x00\x00\x00\x01\x1e\x82\x10\x0c!IhSN: 2130241834\x00\x00\x10\x00\x00\x00\x10\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x1f\x82\x08\x0c!IhA\x1f\x00\x01B\x1f\x00\x00\xfe\xff\xff\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01 \x82\x08\x0c!Ihr#\x00\x00s#\x00\x00t#\x00\x01\xcd#\x00\x00\xd3#\x00\x00\xd4#\x00\x00\xd5#\x00\x00\xd6#\x00\x00\x01 \x82\x08\x0c!Ih\xfe\xff\xff\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00' Ril 8527360 50331649 x -1 bytes
Skipping message. --- Len b'SMA\x00\x00\x04\x02\xa0\x00\x00\x00\x01\x00:\x00\x10`e\x0e\xd0}\x00#\x19\x02#\x00\x01\x8a\x00*\xe9\xf8~\x00\x01\x00\x00\x00\x00\x05\x80\r\x04\xfd\xff\x07\x00\x00\x00\x84\x03\x00\x00\x9d4Ih\x00\x00\x00\x00\xd4\xed\xfd\xfe\xed\xf6\xfb\xfc\xfa\xb9\xb9\xa9\x00\x00\x00\x00' Ril 4797440 2097348609 x -1 bytes
Skipping message. --- Len b'SMA\x00\x00\x04\x02\xa0\x00\x00\x00\x01\x00\xc6\x00\x10`e1\x90}\x00#\x19\x02#\x00\xa0\x8a\x00*\xe9\xf8~\x00\x00\x00\x00\x00\x00\x06\x80\x01\x02\x00X\x00\x00\x00\x00\x03\x00\x00\x00\x01\x1e\x82\x10\x0c!IhSN: 2130241834\x00\x00\x10\x00\x00\x00\x10\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x1f\x82\x08\x0c!IhA\x1f\x00\x01B\x1f\x00\x00\xfe\xff\xff\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01 \x82\x08\x0c!Ihr#\x00\x00s#\x00\x00t#\x00\x01\xcd#\x00\x00\xd3#\x00\x00\xd4#\x00\x00\xd5#\x00\x00\xd6#\x00\x00\x01 \x82\x08\x0c!Ih\xfe\xff\xff\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00' Ril 8527360 50331649 x -1 bytes
Skipping message. --- Len b'SMA\x00\x00\x04\x02\xa0\x00\x00\x00\x01\x00:\x00\x10`e\x0e\xd0}\x00#\x19\x02#\x00\x01\x8a\x00*\xe9\xf8~\x00\x01\x00\x00\x00\x00\x01\x80\r\x04\xfd\xff\x07\x00\x00\x00\x84\x03\x00\x00q7Ih\x00\x00\x00\x00\xd4\xed\xfd\xfe\xed\xf6\xfb\xfc\xfa\xb9\xb9\xa9\x00\x00\x00\x00' Ril 4798208 2097348609 x -1 bytes
Skipping message. --- Len b'SMA\x00\x00\x04\x02\xa0\x00\x00\x00\x01\x00:\x00\x10`e\x0e\xd0}\x00#\x19\x02#\x00\x01\x8a\x00*\xe9\xf8~\x00\x01\x00\x00\x00\x00\x05\x80\r\x04\xfd\xff\x07\x00\x00\x00\x84\x03\x00\x00r7Ih\x00\x00\x00\x00\xd4\xed\xfd\xfe\xed\xf6\xfb\xfc\xfa\xb9\xb9\xa9\x00\x00\x00\x00' Ril 4798208 2097348609 x -1 bytes`

Nilan464, 2025-06-11:
> The integration sensors became unavailable and when I tried to delete and reconfigure I receive this message (see screenshot). [image]
> 
> When I click on add it says "Unknown error occured"

I have this exact same message here when I try to reconfigure it.

Nilan464, 2025-06-11:
Perfect for me with v0.4.10.
The problem is solved.
Thank you very much.

closed 2025-06-11 by littleyoda

Mark481971, 2025-06-11:
Works for me as well. Many thanks!