littleyoda/ha-pysmaplus #80: Request IP Field on Add Integration

steel4me, 2025-09-26:
hallo, vielen dank für diese integration! all meine sma und byd geräte sind in einem vlan und mittels pfsense geschützt. per dhcp werden ip adressen im vlan statisch gesetzt, habe einen Sunny Home Manager 2.0, SUNNY TRIPOWER 10.0 SE und SUNNY TRIPOWER 10.0 und BYD speicher. will den hybrid wr dazu bringen dass er per automatisierung die batterie nicht entlädt wenn meine go-e wallboxen die autos laden.

ich finde keine möglichkeit die ip adress manuell einzugeben, kann das hinzugefügt werden?

littleyoda, 2025-09-26:
I don't understand the question.
For all devices (except SHM/energy meters), you can specify a host name or IP address.

steel4me, 2025-09-26:
Oh oh, I'm sorry, I thought I had to add the home manager, but I can just as well add the hybrid inverter. Does everything work in parallel with the normal SMA integration? He threw a 500 internal server error when adding the SUNNY TRIPOWER 10.0 SE. I have already activated Modbus.

steel4me, 2025-09-26:
Erstmal danke für die schnelle Antwort. Ich habe die WRs bereits über die normale SME Integration für das Energiedashboard hinzugefügt vor etwa 2 Jahren. Das will ich nicht anfassen weil es mir die Stats schonmal zerschossen hat als sich Entitäten geändert haben. 

Dann bin ich per HACS auf deine Integration gestoßen und habe per Suche irgendwo gelesen dass es mit deiner Integration auch möglich sein soll die Ladung / Entladung des Akkus zu steuern. Ich habe nochmal alles entfernt und neu hinzugefügt. Aber so wie ich das sehe muss ich mein Vorhaben trotzdem manuell per Modbus realisieren. Ich dachte ich kann mir diesen schritt sparen.

littleyoda, 2025-09-26:
Die Parameter, die sich aktuell ändern lassen findest du hier:
https://github.com/littleyoda/ha-pysmaplus/blob/main/docs/set_parameter.md
Kannst du genauso gut und ggf. sogar besser, über Modbus lösen.

den anderen Fehler konnte ich gerade nachstellen. Schaue ich mir an. Sollte aber auf die Funktionalität keinen Einfluss haben.

steel4me, 2025-09-27:
Vielen Dank für die Infos und deine tolle Integration.

closed 2025-09-27 by steel4me